FreeCAD/FreeCAD #21928: BIM: Native IFC - Type assignement is not saved in IFC file

semhustej, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Type information is saved to IFC file, but the type is not assigned to instances (occurrences) in native IFC file.

Steps to reproduce:

- open FC file from this zip container

[FC-IFC-type-not-saved.zip](https://github.com/user-attachments/files/20687207/FC-IFC-type-not-saved.zip)

- convert the file to strict IFC mode
- right click the `Door` in project tree  `IFC` -> `Convert to type`
- select to keep the existing object in the dialogue
- the type is created, the `Door` instance has the correct type assigned
- save the IFC file
- the IFC file contains the Type information, but the `Door` instance is not associated with the Type
- I tested viewing the file in Bonsai BIM, BIMvision and re-importing to FreeCAD. All of these confirm the finding.

See behavior in the video:

https://github.com/user-attachments/assets/5b8bde79-8bc6-4461-9624-07635cc31ab1

The saved IFC file is provided in the .zip file.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-06-11:
I can confirm this straight away. It's a known limitation that was here before. Remember the `Classification` traceback we had during IFC object conversion? The same that I've modified misses .ifc file modification. I wanted to do this this week or previous week but I'm swarmed with other issues I want to do first. Will check it if I will find time

semhustej, 2025-06-11:
> I can confirm this straight away. It's a known limitation that was here before. Remember the `Classification` traceback we had during IFC object conversion? The same that I've modified misses .ifc [...]

Thanks for confirmation. This is definitely not a pressing issue and probably on ToDo list of the IFC development. I guess it should be classified as a Feature request.

semhustej, 2025-06-27:
Tested with latest weekly. Works fine, thank you.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.0

```