FreeCAD/FreeCAD #21933: View cube: two buttons does the same action

Catriks, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Expected behaviour: Clicking the arrow rotates the view 90 degrees, clicking the corner rotates the view 45 degrees

Actual behaviour: Both rotate the view 45 degrees

Same behaviour with all directions.
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Finnish/Finland (fi_FI)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Gridfinity 0.11.9
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-06-11:
I disagree: cube corners and edges don't rotate the view but **align** it.
I.e. if you drag-rotate the view randomly and click arrow, the 45° rotation is **added** to the current one but if you click on the cube the rotation is **overwritten**. 

Such behavior is required.

Catriks, 2025-06-11:
> I disagree: cube corners and edges don't rotate the view but **align** it. I.e. if you drag-rotate the view randomly and click arrow, the 45° rotation is **added** to the current one but if you click on the cube the rotation is **overwritten**.
> 
> Such behavior is required.

Fair corretion, even though I dont see in what use case you would want rotate from a random view to exacty 45 degrees to another random view, instead of just using the mouse for rotation. All other CAD I have used use the view cube solely for aligning the view.

Regardless, it is currently not possible to align the view 90 degrees to another side of the cube with a single click, which is often a required action.

Rexbas, 2025-06-11:
There is an option to change the amount of rotation when you press an arrow. 

[image]

If you change `Steps by turn` to 4 then it will rotate in steps of 90 degrees instead of 45 degrees.