FreeCAD/FreeCAD #21937: PartDesign: issues with thickness/chamfer/fillet with compound part

brettowe, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I stumbled across a bug while trying to create an example for someone
if you create 2 sketches, both touching 0,0

[image]
blue highlight is the other sketch

pad 1 normal and in reverse on the other:

[image]
chamfer and fillet won't work on the bottom face of the upper or the top of the lower face
thickness doesn't work on any face

in report view for both of those you'll see "<Exception> TopoShapeExpansion.cpp(3012): Null input shape"

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/80.9812/1
```

### Subproject(s) affected?

None

### Anything else?

example file
[thickness-compound-bug.zip](https://github.com/user-attachments/files/20694965/thickness-compound-bug.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-06-12:
> chamfer and fillet won't work on the bottom face of the upper or the top of the lower face
> thickness doesn't work on any face

Issue can be reproduced as described. Fillet does not work on 4 edges connected to the meeting point. Thickness does not work on any face.

Version:
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.9548/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * sheetmetal 0.7.22
  * Silk 0.1.7

Pendelf, 2025-08-12:
Two non-overlapping extrusions in one body, if you select only one, then the thickness does not apply, if you select both, then everything is fine

https://github.com/user-attachments/assets/edd84f74-d994-4148-a3e8-203fd7296051