FreeCAD/FreeCAD #21942: Part: Attachment not considering Body Placement

Creymore, 2025-06-11:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Dear Dev team,
The attachment engine does not if:
A Body with Sketch, the Body postion is addjusted
Some pictures:

[image]

[image]

[image]

Well it does not do what i expected. it seems like the attachment is placing the Body002 in Global Space, Like the Body is still positioned on (0,0,0) in Global space.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * Dracula 0.0.6
  * fasteners 0.5.38
  * FreeCAD-themes 2025.1.7
  * freecad.gears 1.3.0
  * FusedFilamentDesign 0.25.220
  * Ondsel-Lens 2024.11.29.01 (Disabled)
  * OpenTheme 2025.5.20
  * ProDarkThemePreferencePack 1.0.0
  * SaveAndRestore 0.2.1
  * sheetmetal 0.7.22
  * Telemetry 1.0.1
```

### Subproject(s) affected?

None

### Anything else?

I would have loved to upload the file but it did not let me :(

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-06-11:
Confirmed
Body001 is attached to the sketch edge without offset but doesn't belong to it if the sketch is moved by its owner.
[image]

[AttachementBug.FCStd.remove.zip](https://github.com/user-attachments/files/20697957/AttachementBug.FCStd.remove.zip)

IMHO this feature shouldn't change because it will break old files.
The only solution I see could be to add a "Use Global Placement" boolean property to the AttachementExtension so it could switch to a new behavior which fixes this issue.

kaiwas, 2025-06-12:
> "Use Global Placement"

This would be a very good opportunity.

kadet1090, 2025-08-13:
@FlachyJoe Will you be interested in implementing that kind of solution?