FreeCAD/FreeCAD #21946: Unfriendly UI performance leads to failure and interest into disinterest.

DaemonDave, 2025-06-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi:  I'm Dave. 
I've reviewed over a thousand open source projects from academia to company to solo inventors. 
I wrote a book about software reuse called 
https://www.amazon.ca/Untrapped-Value-Software-Powering-Prosperity-ebook/dp/B0B4F7Y2NP

I've fiddled with tutorials and loading models with freecad 1.0 and 1.01 and I need you to understand: unless you prioritize explaining and following "work flows" for neophytes attempting to pick up and just use your free software it is doomed to the dustbin as your new users get fed up and leave. I am at the precipice myself. OpenSCAD by contrast takes little to understand and operate not expertly but basically. You have no excuse. 

I've predicted the downfall of many sources and yours won't be the next linux, the most successful software project in history. Unless you accept the same wisdom they under Linus used. 

You don't make software for others if you make them fed up and quit. That's your support system for the next generation.  We didn't write your stupid code nor were we in your headspace so if you don't care or understand why people can't make your code work it's all on you and your ego. 

All software projects is a community first and software last. Linus used to make major changes asked for because it was faster and easier. That drew MORE SUPPORT than asking others to do it. How's YOUR user count faring?  Dropping off? 

Windows that don't open clearly and obfuscated class associations ( why do i open a part but need a sketch but then i can't make a poly line but I can't) because your design is so awesome and only fools don't understand is a symptom of lonely introverts that can't accept their code is shit. If i cannot use it, it's shit. 

Your unwillingness to accept you're wrong is the obstacle not the new user. Sorry, but I've seen what happens to you a thousand times. We neophytes have M.Sc.s and Ph.D.s also don't insult us with your ignorance. 

Freecad is now shit. 

If you think I don't know code, I wrote my first BASIC program in 1987 on a Commodore, I've written ASM, COBOL, FORTRAN, C, C++, ADA,  etc...  all before 2000. 

Here's my hard truth to most OOP devs, when you get lazy on object design and adoption you cause the confusion problems that waste most of your time trying to resolve. If you don't follow the unix design philosophy,  you make it worse. 

I am willing to present a real reasonable offer redesign plan to you for free that will make what's actually important as the critical path which i believe will fast track user satisfaction which should draw MORE devs which should alow you to get more help faster.  I'm not asking for anything but priority to the things devs think are beneath them but actually cause more improvement than a new feature. 

If wanted, please contact me. If not, I'll check back if youre still around. 

### Full version info

```shell
Freecad 1.0 and 1.01
```

### Subproject(s) affected?

None

### Anything else?

Heres the cheaper second book version...
https://play.google.com/store/books/details/Untrapped_Value_1_01_Software_Reuse_Powering_Futur

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

kaiwas, 2025-06-12:
Very nasty, blatant advertising.
I read the text and felt like I stepped in shit.

maxwxyz, 2025-06-12:
Hi Dave, you can join our Discord and cooperate with the Design Working Group.

closed 2025-06-12 by maxwxyz

DaemonDave, 2025-06-12:
This:

_"Very nasty, blatant advertising.
I read the text and felt like I stepped in shit.

You have the opportunity to help, but it is definitely not done with the help of such messages."_

is TEXTBOOK Developer Superiority Syndrome:

**"Look at me, I'm so smart, if you don't understand how to use my code it's all on your stupidity, stupid operator."**

If you are a lonely introvert and you think people should be in awe of your dev skills and they need to bow to you then I have seen thousands of you while you think you're special; turns out - _you're not_. You're a common problem. Not my first lonely introvert, not my first valuable offer to assist your success. 

**I can count the majority of over 1000 open source code project are occupied by people that need to get over themselves before they have a successful project.**

**BECAUSE IF YOU DON'T THEN YOU GET TIRED OF GIVING TIME TO A POINTLESS PROJECT WITHOUT ACCOLADES OR RECOGNITION AND THEN YOU GIVE UP AND GO AWAY**

That's on top of the natural turnover from students getting work, unemployed getting a job, and so on. Natural life circumstances affect and effect all devs, including Linux. Why doesn't Linux die? I know. Do you?

It's an indicator of no critical-introspection to feel like it's the **complainer's** fault. Does anyone actually want to spend time complaining about code that doesn't work? I wanted to be finished a mounting bracket in under a week. My outcome: freecad isn't well designed enough to make that clean and clear.  Kaiwas wasted **MY** time first.

Gonna cry more?

Kaiwas didn't stop to think before he blurted out and exposed himself. Do you think it's a good idea to listen to people like him? **My research says he's the REASON your project will probably fail.** 

**I offered to present ideas for FREE based on years of experience - that I had an obvious way to prove it - to assist while pointing out with tough love why I don't see a prosperous future. His response?  SHOOT THE MESSENGER.**

Which one of us is the moralizing scold and which is the honest broker? See how easy it was to outflank you, Kaiwas? Why? 

Thanks for proving my point: I'm not blatantly advertising anything, but you are making it clear you know FAR LESS about what I know than I do.

See, here's how one helps without becoming slave to other dev's neediness, I'm too busy to join your channel and waste my time, and you aren't paying me, so rather I will present what I mean in the included powerpoint and I will happily answer SPECIFIC TARGETTED questions, within reason. 

[Freecad1.pdf](https://github.com/user-attachments/files/20716480/Freecad1.pdf)