FreeCAD/FreeCAD #21958: BIM: Wall `move base`  property doesn't work if you enter X,Y,Z value from keyboard

ErtugrulEge, 2025-06-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Greetings everyon!

When you create a wall, if you change X,Y or Z position from keyboard `move base` property doesn't work but if you use move tool base line moves with the wall.

For example, create a wall. Default location Z is 0 then change to 3mt you will see base will not move with wall  but if you move wall with draft move tool the  wall base as expected will move also.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

semhustej, 2025-06-12:
I think this is because if you use the Draft_Move tool, it is moving the base Line object (as set in the Wall properties).
When you change the placement parameter of the wall (Z value), you are affecting the Wall object. That is why the base Line object is not moving.

At least this is my understanding.

furgo16, 2025-06-13:
> UPD/ I will try to solve the problem using expressions. And analyze possible mutual changes in the coordinates between parents and descendants. It will take time, but I feel that it needs to be done)

I've tried this, and ended up with circular dependencies errors/warnings. It's been a while since I did this, so I don't have the actual example at hand, sorry. You have more experience than me here, so you might succeed where I failed.

kaiwas, 2025-06-13:
> I've tried this, and ended up with circular dependencies errors/warnings. It's been a while since I did this, so I don't have the actual example at hand, sorry. You have more experience than me here, so you might succeed where I failed.

Cyclic dependencies are not really such. It's just about the direction of data transfer.
For some reason, data sent from a child to a parent is always cyclic.
But this can be solved with the help of the hraf() wrapper

For example, with the help of such a wrapper, you can change the height of a wall with a window using the height of the window as a reference point. By default, this is marked as a "cyclic dependency", but in fact, the height of the wall does not affect the height of the window. Therefore, this is not really a cyclic dependency.

kaiwas, 2025-06-13:
Pay attention to what happens if you move a wall with the "Move base"=true setting. The base moves. It acquires new coordinates.
At the same time, the coordinates of the wall itself remain unchanged )

That is, with this setting we move the base, and the wall is rebuilt after.
If you make "Move base"=true, then the object itself moves. That is, the wall changes coordinates.

Oh.... How complicated it all is )

https://github.com/user-attachments/assets/02c89a52-3bd7-43fb-9cfa-859af280991b

My opinion is that you need to "glue" the coordinates of the base and the object. And these two properties ("Move Base" and "Move with host") should be removed completely, since they will then become unnecessary. They do nothing but complicate things.

markkomartin, 2025-06-13:
> My opinion is that you need to "glue" the coordinates of the base and the object. And these two properties ("Move Base" and "Move with host") should be removed completely, since they will then become unnecessary. They do nothing but complicate things.

When baseless walls will be available we will always use those and we will not have this problem anymore. 
However, walls with a base will also remain. 
Yorik mentioned some cases where the base needs to be separated from the wall. 
If we paste the base and wall coordinates it could be a problem for these cases.
I can't imagine what cases that might be at the moment

ErtugrulEge, 2025-06-13:
In fact, if you glue wall and base you will get  a kind of baseless wall.What I understand in this discussion ,we need to simplify things.

paullee0, 2025-06-29:
Move Base behaviour is not good for Clones and Links :

1.  https://github.com/FreeCAD/FreeCAD/issues/22162#issuecomment-3016348435
2.  https://github.com/FreeCAD/FreeCAD/pull/21005#issuecomment-3015185712

zbynekwinkler, 2025-06-29:
> In fact, the current system needs to be simplified. Now it is very complex and any movements cause the model to fall apart. That is, it is possible to do something once (I mean modeling), but changes are very difficult.

I can related to this. Doing any change after the initial modeling feels like entering a minefield (for a beginner).

Would it make sense to use a similar concept to `Part` and `PartDesign`? There is no _host_ there. Instead there is [Std_Part](https://wiki.freecad.org/Std_Part) and [Part_CoordinateSystem](https://wiki.freecad.org/Part_CoordinateSystem). Maybe `Site`, `House`, `Level` and `Wall` could be `Std_Part` objects with LCS. I couldn't find anything similar to `Move Base` in Part or PartDesign so maybe it wouldn't be needed.