FreeCAD/FreeCAD #21962: C++ DXF Importer - import Blocks

semhustej, 2025-06-12:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request**

**Current state**
Currently C++ DXF Importer does not import blocks. There is an option to 'Group Layers into Blocks' which creates compound (note: not Part_Compound, but OCC Compound object as described here: https://github.com/FreeCAD/FreeCAD/issues/7551#issuecomment-1479287042). But this is not adequate, as it does not follow how blocks are utilized in DXF.
Currently C++ Importer explodes the blocks and imports them as a simple geometry.

**Feature request**
Import DXF Block objects as a compound objects using C++ DXF Importer. I think there are two possible options:

1. Import DXF Blocks as a Part_Compound objects (with the content of the block inside of the compound). https://wiki.freecad.org/Part_Compound
2. Import DXF Blocks as OCC Compound objects (as the Python importer does)

The Legacy Python DXF importer imports DXF Blocks as OCC Compounds.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42133 (Git)
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

This issue was create to log a missing C++ Importer feature compared to Python Importer.

Not sure if it is needed, but here are two test dxf files.

[DXF-sample-files.zip](https://github.com/user-attachments/files/20712544/DXF-sample-files.zip)

1. simple dxf file with three blocks created differently in relations to placement of objects on layers

[image]

2. simple dxf architectural drawing file with few blocks inside

[image] 

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

furgo16, 2025-06-15:
@semhustej would the attached FreeCAD file be a viable result for implementing this feature?

- BLOCK definitions as objects in a hidden group
- INSERT instances as `App::Link`s with `LinkedObject` property set to the original BLOCK object

[FC-DXF-block-testing.zip](https://github.com/user-attachments/files/20746781/FC-DXF-block-testing.zip)

furgo16, 2025-06-15:
> The files do not contain multiple instances of one block definition. Only one instance is placed in the drawing.

That's correct, and that's what that output provides given the DXF file. It has 3 different blocks defined, and each block is inserted once. Opening the file in QCAD, I see the three instances, the block definitions are not accessible (or at least I could not find them, I'm not well-versed in QCAD). With the WIP branch I'm putting together, in FreeCAD you get the block definitions as hidden objects, and the insertions as links to the original block definition insert. You can have one or multiple inserts.

 If I interpret the DXF standard correctly, BLOCKs are the definitions and INSERTs are the actual instances. The idea being that the BLOCK is defined only once, but it can be instanced multiple times (which is when INSERTs are used). This file simply happened to have one instance of each block.

furgo16, 2025-06-16:
Ok, that seemed to work as expected. Here's the FreeCAD file after the import: [block-multiple-instances.zip](https://github.com/user-attachments/files/20757262/block-multiple-instances.zip) (uncompress before opening).

```
--- DXF Import Summary ---
DXF Version: AC1027
File Encoding: utf_8
File Units: Millimeters (from $INSUNITS)
Manual Scaling Factor: 1.000000
Final Scaling: 1 DXF unit = 1.0000 mm

Performance:
  - C++ Import Time: 0.3667 seconds
  - Total Import Time: 0.5391 seconds

Import Settings:
  - Import hidden blocks: No
  - Import layout objects: No
  - Import points: No
  - Import texts and dimensions: No
  - Join geometry: No
  - Manual scaling factor: 1.000000
  - Merge option: Create Part shapes
  - Use colors from the DXF file: Yes
  - Use layers: Yes

Entity Counts:
  - BLOCK: 1
  - CIRCLE: 1
  - INSERT: 6
  - LWPOLYLINE: 4
----------------------------
  Total entities read: 12
FreeCAD objects created: 9

System Blocks:
  - Other System Blocks: 3

Unsupported Features:
  (None)
--- End of Summary ---
```

semhustej, 2025-06-16:
> Thanks for the correction, that's were input from someone with experience in DWG/DXF editors is so valuable, even if in the end we reached the same conclusion.

No problem.

I guess the user will still have option not to import blocks and in that case  the base geometry of exploded blocks is imported on layers on which it was inside of the block.

semhustej, 2025-06-18:
> I've started documenting use cases and concepts. It's essentially the "Compound object workflow" at https://wiki.freecad.org/DXF#BLOCK_placement_strategies

I would add these texts in these positions to make the relation to the original DXF file objects clear:
[image]

A: This is the block definition in from the imported DXF file.

B: These are block references from the imported DXF file.

From a user perspective, the terms `BLOCK` and `INSERT` are not used, instead `Block Definition` and `Block Reference` are used (as you can see in the user facing AutoCAD documentation from the previous post)

furgo16, 2025-06-18:
> Such thing as 'Standalone block' does not exist.
>When you join a geometry in DXF into block, the block definition is created. Than block instances can be placed in the drawing.
> It is also possible to not have any block instances in the drawing. Only block definition than exist in the DXF file.

Understood, thanks. The later is what I was refering to when I mentioned standalone blocks => unreferenced, uninstantiated blocks. But I understand that they're not supposed to be seen or interpreted as part of the drawing. From what you're saying, they're either leftovers (folks might forget about them because they're not visible?) or for future use as a template.

I noticed QCAD even has a command to "Purge unused blocks".

> So we could think of importing Block definitions, which are in the DXF file, but do not have any instance placed in the drawing. I would not complicate things and would not import such blocks (not sure how it's handled currently). I have encountered many DWG/DXF files which have a lot of blocks, which are not used in the drawing and would make the import process slower with not benefit for the user.

Let's not worry about the implementation (the linked branch already imports all blocks, so that work is already done). Let's try to agree on the principles for importing unreferenced blocks. 

 Thinking about it, there might be two opposite ways of considering them:

1. Clutter that we do not want to import
2. Data that needs to be kept

If our priority would be 1, we can just stop importing unused blocks. If our priority would be 2, we can import them all unconditionally.

A third, hybrid approach would be to let the user decide. We have too many settings already, so an option might be to repurpose the "Import *blocks" to be "Import hidden blocks" and that be the switch that controls whether both anonymous and unreferenced blocks are imported or not.

Thoughts?

closed 2025-07-12 by yorikvanhavre