FreeCAD/FreeCAD #21974: Preferences: Freeze when clicking off color picker

ryankembrey, 2025-06-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When choosing a color in preferences, if you open a color picker, then click off the color picker window, FreeCAD freezes.

### Full version info

```shell
OS: Fedora Linux 41 (Sway) (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.42136 +1 (Git)
Build date: 2025/06/10 11:30:57
Build type: Release
Branch: add-spacing-preview
Hash: 1d23255f1f27296db8c16cbbb322556d68dd28f4
Python 3.13.3, Qt 6.8.2, Coin 4.0.2, Vtk 9.2.6, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.5, OCC 7.8.0
Locale: English/Australia (en_AU)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/159.34/1
Installed mods: 
  * OpenTheme 2025.5.20
  * CfdOF 1.30.5
  * sheetmetal 0.7.22
  * fasteners 0.5.38
  * Render 2024.12.15
  * Telemetry 1.0.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-06-14:
I can't reproduce.

<details>

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42080 (Git)
Build date: 2025/06/05 12:39:32
Build type: Unknown
Branch: master
Hash: 5c219bea4187030d56c998bc318bdf4869209ced
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

```ruby
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42006 (Git)
Build date: 2025/06/02 16:32:08
Build type: Release
Branch: main
Hash: db1b5a877314f8d21514680421649fd9dc65acde
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/102.706/1
```

</details>

@ryankembrey, can you reproduce the freeze with stable version and/or AppImage?

ryankembrey, 2025-06-15:
@xtemp09 
I can reproduce on the 1.0.1 AppImage on SwayWM.

I can not reproduce on Gnome Wayland session or Gnome Xorg session. 

So, there seems to be a problem with how SwayWM is focusing the windows. When on sway, after I click off the color picker window, the color picker window disappears but my desktop shows an alert from the current desktop number I'm on (which usually happens when a window or tab opens in a desktop number that I'm not on).

I'm not sure whether this is the fault of SwayWM or FreeCAD

xtemp09, 2025-06-15:
@ryankembrey, how many screens do you have?

ryankembrey, 2025-06-15:
@xtemp09 Just the one moniter.

ryankembrey, 2025-06-15:
@xtemp09 I am not sure what you mean by activate/deactivate.

When going into preferences menu and selecting a file picker, if i click off it the same freeze happens. But when accessing a file picker without having preferences open, clicking off it does not freeze FreeCAD.

I think the issue may be that the color picker modal dialog is on top of another modal dialog (preferences window), i.e. stacked modal dialogs.

MisterMakerNL, 2025-06-17:
This can be an opentheme issue, as I had this issue a while back in default stylesheets. Please test without stylesheets or with default.

luzpaz, 2025-10-05:
A [Heisenbug](https://en.wikipedia.org/wiki/Heisenbug)!

Lets close this and if you can reproduce again, just give us a heads up.

closed 2025-10-05 by luzpaz