FreeCAD/FreeCAD #21977: BIM: Remove unnecessary "Move from host" setting on all architectural components (except windows and doors).

kaiwas, 2025-06-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At the moment, the program has a setting "Move with host", which is present on all BIM objects. But in fact, it affects ONLY windows and doors. Because only windows and doors have "Host".

In general, this does not affect the process, but can be a cause of confusion, because "Host" can be perceived as "Base" and vice versa.

The screenshot shows the properties of the beam. You will not find the Host, although "Move with host" is there.

[image]

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42133 (Git) AppImage
Build date: 2025/06/09 17:51:59
Build type: Release
Branch: main
Hash: 12b4610b499a9756f19bbb85f59fe16e5fa135db
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.63
  * freecad.gears 1.3.0
  * sheetmetal 0.7.22
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

markkomartin, 2025-06-13:
I agree. 
Thanks @kaiwas for reporting.

ErtugrulEge, 2025-06-13:
I don't agree.If you use BIM objects in a `level `and if you want to change for example height of the` level `all object will move with `level`. A` level `is a host for BIM objects, so when `move with host=yes` I can move all objects with level. 

That is useful, but I agree there are confusing things.We should clear what is a host? for example a wall is a host  for door and window, if I move wall I can move doors and windows with wall. At this time this is not working. 

We souldn't remove `move with host property `instead We should fix this kind of issues
https://github.com/FreeCAD/FreeCAD/issues/21439#issuecomment-2970040284 

[image]

kaiwas, 2025-06-13:
> .If you use BIM objects in a `level `and if you want to change for example height of the`level`all object will move with `level`. A`level`is a host for BIM objects, so when `move with host=yes` We can move all objects with level.

Hmm. Indeed.
But then you need to tidy up the properties. Add "Host" property to all objects where there are none now.
Because there is no mention on the wall of the level it is located on.

maxwxyz, 2025-06-13:
@FreeCAD/design-working-group FYI

markkomartin, 2025-06-14:
> I don't agree.If you use BIM objects in a `level `and if you want to change for example height of the`level`all object will move with `level`. A`level`is a host for BIM objects, so when `move with host=yes` We can move all objects with level.
> 
> That is useful, but I agree there are confusing things.For example, beam should have a `host `property when I put in a ` level` that should be ` Host=level` . We should clear what is a host? for example a wall is a host for door and window, if I move wall I can move doors and windows with wall. At this time this is not working and `level `is a host for wall.
> 
> We souldn't remove `move with host property ` instead We should fix this kind of issues [#21439 (comment)](https://github.com/FreeCAD/FreeCAD/issues/21439#issuecomment-2970040284)
> 

Thanks @ErtugrulEge for this message. 
I didn't know that the wall became a guest of a level. 
Since there is no host = level property I thought it was using another method.
If we can do what you say, it all starts to make a lot of sense.

ErtugrulEge, 2025-06-14:
To clear the confusion:
1. Wall is a host for doors and windows.Doors and windows should follow their host wall
2. Level is a host for wall and other BIM objects. Wall and other objects should follow their host level

as you see doors and windows are exception for Level object

paullee0, 2025-06-29:
My suggestion is discussed here - https://github.com/FreeCAD/FreeCAD/pull/21005#issuecomment-3015185712