FreeCAD/FreeCAD #21979: BIM: Site WindRose does not scale along with Solar Diagram

Harvie, 2025-06-13:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When i scale up the solar diagram, the wind rose stays small.
(When i move it, they move together tho...)

[image]

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git)
Build type: Release
Branch: makepkg
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Harvie, 2025-10-16:
Yeah, i've found the issue https://github.com/FreeCAD/FreeCAD-Bundle/pull/412

luzpaz, 2026-01-16:
@Harvie Is this solved by adding the ladybug dependency ?

Harvie, 2026-01-16:
Eeeh. No. Mixed up two issues. Sorry.
ladybug is needed to reproduce this. but does not solve the problem.

FEA-eng, 2026-04-17:
I assume it's still relevant, but maybe @Harvie could reconfirm.

semhustej, 2026-04-17:
I am able to reproduce the issue.

```
OS: Debian GNU/Linux 13 (trixie) (GNOME/gnome-xorg/xcb)
Architecture: x86_64
Version: 1.2.0dev.20260407 (Git shallow) AppImage
Build date: 2026/04/07 22:56:13
Build type: Release
Branch: (HEAD detached at 6db886fe0)
Hash: 6db886fe0e79bf060ffeaa50198da25702a17e31
Python 3.11.14, Qt 6.8.3, Coin 4.0.8, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Ireland (en_IE)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1

```

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/cb57730da797b57ff3d29154e7b6e7ff0f82dceb

The wind rose gets rebuilt when the solar diagram scale changes. Could be picked up for a PR.