FreeCAD/FreeCAD #21984: Sketcher: Angle Equality Constraint

leoheck, 2025-06-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Can we have a constraint to make angles equal?
Alternatively, options for complementary and supplementary would make it complete.

### Full version info

```shell
NA
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-07-10:
It wasn't possible due to https://github.com/FreeCAD/FreeCAD/issues/22034
But when that is solved then we can have constraint that have more than 3 references. So that should be possible

PaddleStroke, 2025-11-17:
yes indeed.

leoheck, 2025-11-17:
A quick and nasty implementation could be to add a name to the 3rd parameter (angle) if it does not have it already, and then create the new angle constraint using that name. =)

wt, 2026-09-07:
Would love to see this implemented. Would have been useful in a number of models I have made.