FreeCAD/FreeCAD #21986: CAM: Class and asset folder terminology cleanup proposal

knipknap, 2025-06-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

While working on the tool model, I found (and admittedly also introduced) some inconsistencies in terminology.

- The folder that contains the Bit, Shape, and Library folders is sometimes referred to as "Tools" folder, sometimes "Asset", and due to #21775 the default config folder is now named "CamAssets". I propose to rename `src/Mod/CAM/Tools` to `src/Mod/CAM/CamAssets`, and align all mentions of the folder in the code base accordingly. I think it provides more clarity for users and developers alike.

- Asset classes, asset names, and folders are also inconsistent. Here is an overview of the status quo:

  Class Name          | Asset Name (URL scheme) | Folder name
  ------------------- | ----------------------- | -----------
  Library             | library                 | Library
  ToolBit             | toolbit                 | Bit
  ToolBitShape        | toolbitshape            | Shape
  ToolBitShapeSvgIcon | toolbitshapesvg         | Shape
  ToolBitShapePngIcon | toolbitshapepng         | Shape
  
  Currently the fact that icons are stored in the same folder as the ToolBitShape asset type also has a negative impact on performance; I had to implement an inefficient bi-directional Regex based mapping from asset name to folder. This could be cleaned up by consistent naming and a "one asset type per folder" policy.
  In addition, in the future there may be more asset types: machines, spindles, chucks, fixtures, and potentially many more.
  
  For long term consistency, I propose to adopt a consistent naming scheme:
  
  Class Name          | Asset Name (URL scheme) | Folder name
  ------------------- | ----------------------- | -----------
  Library             | library                 | Library
  Bit                   | bit                 | Bit
  BitModel        | bit-model            | BitModel
  BitModelSvg | bit-model-svg         | BitModelSvg
  BitModelPng | bit-model-png         | BitModelPng
  Machine             | machine         | Machine
  MachineModel        | machine-model    | MachineModel
  Spindle             | spindle         | Spindle
  SpindleModel        | spindle-model    | SpindleModel
  ChuckModel          | chuck-model      | Chuck
  Fixture             | fixture         | Fixture
  FixtureModel        | fixture-model    | FixtureModel

Advantages:
- Consistency.
- Clarity of purpose: E.g. all Model classes directly represent FreeCAD files, all other assets do not.
- Maintainability & opportunity for deduplication: Asset names can be mapped generically, and the Model classes can even share a common base class that handles most of the code.
- The consistent naming and *one folder for one asset type* policy allows for a simpler and more efficient implementation of the [FileStore](https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/CAM/Path/Tool/assets/store/filestore.py).

Migration of existing asset folders to to new structure can easily be done by an automatic migration function that looks for existing old assets and moves the to them new ones.

The only drawback I see is that I am starting to worry about stepping on toes because of all the changes I am introducing. But if the above is seen as the right thing to do, then I can prepare a PR.

### Full version info

```shell
OS: Ubuntu 24.04.2 LTS (Unity/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.42143 (Git)
Build date: 2025/06/13 16:54:54
Build type: Unknown
Branch: early-prefs
Hash: ecfb60d003b20e58819742c00bb31cfc531f81e2
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/139.67/1
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

knipknap, 2025-06-15:
@maxwxyz I think this is unrelated to FEM

jffmichi, 2025-06-15:
I don't think it is nice to have the pngs and svgs in different folders. Those are all icons/image and having to put them in different folders only based on their extension is somewhat pointless (imho). How about a folder called "Icons" for all of them. At least something more descriptive to the use of those files and not the extension, maybe like "BitModelIcon". Maybe similarly for a common folder for "Models" (not sure if it makes sense here)?

Connor9220, 2025-10-16:
As of now, here is the state of things.

The Assets directory now defaults to CAMAssets
The Tools directory is now under that specified folder.  
Under Tools is Bit / Library / Shape 

| Class Name | Asset Name (URL scheme) | Folder name |
|--------|--------|--------|
| ToolBit | toolbit | Tools/Bit |
| ToolBitShape | toolbitshape | Tools/Shape |
| ToolBitShapeSvgIcon | toolbitshapesvg | Tools/Shape |
| ToolBitShapePngIcon | toolbitshapepng | Tools/Shape |
| Library | library | Tools/Library | 

Job Templates can now be in the legacy root, or in CAMAssets/Templates

Going forward, we can continue to do stuff like Machine (for json files) and MachineModel or MachineShape (for models) under the CAMAssets file.

One last note:  We've introduced versioning, So, when the user starts up a new version of FreeCAD, it will prompt if he wan'ts to migrate his data.. if so for version 1.1 it will create a v1-1 under the CAMAssets directory.

Here is an example of my setup. 

/home/user//Documents/FreeCAD/  
/home/user//Documents/FreeCAD/Tools
/home/user//Documents/FreeCAD/Tools/Bit
/home/user//Documents/FreeCAD/Tools/Library
/home/user//Documents/FreeCAD/Tools/Shape
/home/user//Documents/FreeCAD/CAMAssets/
/home/user//Documents/FreeCAD/CAMAssets/v1-1/Tools
/home/user//Documents/FreeCAD/CAMAssets/v1-1/Tools/Bit
/home/user//Documents/FreeCAD/CAMAssets/v1-1/Tools/Library
/home/user//Documents/FreeCAD/CAMAssets/v1-1/Tools/Shape

I did not let it migrate my files as it would have copied everything in /home/user//Documents/FreeCAD/  into the new /home/user//Documents/FreeCAD/CAMAssets/v1-1/ location.  I manually migrated my tool data.  As we add new versions

IF you don't migrate, AND FreeCAD doesn't find a v1.1 (or whatever version your using), it will default to the original location.

sliptonic, 2025-12-03:
ok.  Then can we close this issue?