FreeCAD/FreeCAD #21987: PartDesign: Loft creates intersecting surfaces

ElementW, 2025-06-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Maybe a https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20OCC bug, maybe not.

Using an [image] *AdditiveLoft* on 2 rings (2 circles each in Sketcher) *can* create flipped, intersecting surfaces between the 2 profiles.

[image]

Whether it happens depends on some hidden, arbitrary, internal ordering of primitives in the sketches, or their transforms thereof (in the example file, a Pad). While a functional workaround, fiddling with the sketches (usually means rebuilding the geometry, flipping constraints etc) until the result is correct is not acceptable UX.

### Full version info

```shell
OS: Void Linux (KDE/wayland)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git) AppImage
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.38
  * FreeCAD-Ribbon 1.9.0.1
  * SearchBar 1.0.2
```

### Subproject(s) affected?

PartDesign

### Anything else?

[its-flipped-boi.FCStd.zip](https://github.com/user-attachments/files/20739135/its-flipped-boi.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-14:
Seems a duplicate of #6130.

ElementW, 2025-06-14:
Ah, indeed. Not a single mention of "intersect(ing)" in that issue, no wonder I didn't find it.