FreeCAD/FreeCAD #21993: sketcher    -        add separate menus for auxiliary geometry and auxiliary constraints

free777cat, 2025-06-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The essence of the proposed changes (leaving it as it is now for users who want to switch menus) is to add the ability to display two separate menus for auxiliary geometry and auxiliary constraints without having to switch.

### Full version info

```shell
[code]
OS: Linux Mint 22.2 (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * freecad.gears 1.3.0
  * Assembly4 0.50.6
  * fasteners 0.4.72
  * sheetmetal.zip
  * lattice2 1.0.0
  * sheetmetal_V0.6.02.zip

[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

free777cat, 2025-06-15:
[image]

[image]

pierreporte, 2025-06-15:
Alibre has a separate menu for construction geometry, but it’s a single drop-down button.

[image]

I’m not sure having the same mechanism for references would be that useful though. Reference dimensions are much less common. It would be better to have a context menu that lets you select the mode while applying a dimension, like in Catia, as requested in #11413.

Ping @FreeCAD/design-working-group

free777cat, 2025-06-15:
> Alibre has a separate menu for construction geometry, but it’s a single drop-down button.
>
> [image]
>
> I’m not sure having the same mechanism for references would be that useful though. Reference [...]

Sorry, but you have in mind a completely different approach to the process.
Below I will show an example of a menu that I find very convenient to use. If you add a menu of auxiliary geometry and auxiliary dimensions in the same style, it will remove unnecessary clicks from the workflow and make it more "smooth".
Users who wish to use the current process organization do not risk anything.

free777cat, 2025-06-15:
[image]

free777cat, 2025-06-15:
I had in mind that these menu bars could be added as desired.

FlachyJoe, 2025-06-15:
<strike>menu</strike> **toolbar**

free777cat, 2025-06-16:
> menu **toolbar**

Thank you. It's really a toolbar, not a menu.

luzpaz, 2025-06-16:
> Version: 2024.1006.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20241003stable)
Hash: 8058dcf52c2661edabf36d8147d5b8a187378d2d
Python 3.11.9, Qt 5.15.13, Coin 4.0.1, Vtk 9.2.6, OCC 7.7.2

Any chance you can run the latest weekly (which also is based on Qt6) ?

free777cat, 2025-06-16:
I could probably run this in a live linux iso

free777cat, 2025-06-16:
This will be a real convenience for many users. What I am suggesting can be enabled in the settings as a choice between the current way of controlling and a more advanced display of capabilities. 
When designing mechanical parts using the graphical modeling method, you constantly need to switch between geometry modes. Sometimes it's almost a "checkerboard" process.
Using a panel that presents all the tools allows you to focus on the design process, rather than being distracted by thinking about what you need to do/switch to get the line or dimensions you need right now.
I tried adding such a fake toolbar to Sketcher and modeled the work - it turned out to be a significant simplification of the process itself, due to the visually available information of direct action.

obelisk79, 2025-06-16:
Ping @FreeCAD/cad-working-group opinions? Does such an option/feature make sense from your point of view? From a UX perspective I disagree with the premise here.

free777cat, 2025-06-17:
[https://x.com/S_t_a_s_e_k/status/1934764069481009521](url)

Here I opened the latest posts about FREECAD ....

[image]

" I don't like freeCAD, it's not user-friendly, I use Fusion 360 and am considering purchasing a standalone license for SolidWorks. "

These are the impressions of users.

kadet1090, 2025-06-17:
@free777cat Fusion 360 does not have the feature you are asking for, at least as far as I am aware. Each and every user has different imagination of how the software should work.

As for your feature request - from the defaults point of view it does not make sense - too verbose for most users probably. That said, as additional, custom toolbar, that you could enable - it probably could be implemented and useful. What @maxwxyz suggested is so you don't have the toolbar enabled by default but can right-click on the toolbar area and through customization enable these additional toolbars.

Problem here comes from implementation - essentially it would require for each command to be duplicated two times: 1 - the version depending on context, 2 - true geometry only version, 3 - construction geometry only version. That would add additional maintenance burden and it is hard to say if the cost/value factor is good here. If someone finds a good way to implement it without needing code duplication (which may be absolutely possible, have not checked too deeply) and it is implemented in a way that it is opt-in - then it should not be a problem to provide such feature.