FreeCAD/FreeCAD #21996: "Dark theme + Dark background" doesn't make background dark

dmchmk, 2025-06-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi! Clean installment. In the initial configuration dialog I chose Light theme but later decided to check Dark one and got this:

[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-15:
Have you tried restarting FreeCAD?

dmchmk, 2025-06-15:
> Have you tried restarting FreeCAD?

Yes, it stays the same

closed 2025-06-16 by maxwxyz