FreeCAD/FreeCAD #21997: FreeCAD 1.0.1 flatpak uses 2% cpu at idle

peterwwillis, 2025-06-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When I start up FreeCAD and load a model I'm working on. It defaults to the "Part Design" workbench. I then switch to a different workspace and do something else.

I check `top` and FreeCAD is hovering around 2% CPU use even though I'm not using it at the moment.

I ran strace on the process and it's just running a ton of ppoll() calls. Attached is the log (from running `( date ; (strace -f -F -p318490) & sleep 10 ; killall strace ; date ) 2>&1 | tee foo.log`)

[foo.log](https://github.com/user-attachments/files/20746794/foo.log)

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 878f0b8)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.12.10, Qt 6.9.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * Assembly4 0.50.18
  * Curves 0.6.63
  * A2plus 0.4.68
  * ExplodedAssembly
  * woodworking 1.0.1.39285.1.1.1.2025.06.11
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-16:
@adrianinsaval is this flatpak related?

hyarion, 2025-06-16:
It's around 1.3% at idle for me on macos, both 1.0.0 and latest main in a debug build.

[image]

It looks to me like it is due to input polling in qt and coin.
We might be able to change polling rate when the program is not in focus. That could save up around 1/4 of the usage. I haven't tested that though.

luzpaz, 2025-06-16:
@hyarion what too are you using to profile the memory ?

luzpaz, 2025-12-24:
@peterwwillis Any chance you can test the AppImage to see if you can reproduce on the latest 1.1RC1 ?  https://github.com/FreeCAD/FreeCAD/releases/tag/1.1rc1

luzpaz, 2025-12-30:
> It looks to me like it is due to input polling in qt and coin.
We might be able to change polling rate when the program is not in focus. That could save up a few precent of the usage. I haven't tested that though.

Any devs interested in tackling this ?