FreeCAD/FreeCAD #22010: Preferences | Display | 3D View | Marker Size:   Broken?

garyrussell, 2025-06-16:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I just downloaded the latest weekly (42149 for MacOS x86) and this setting seems to be broken. I use large markers (points 25x) but when I use this version, the markers (points) are tiny. I tried changing the setting but saw no changes in the sketches; it's very difficult to select these tiny points when adding constraints.

I have an (much) older weekly on another computer and it has no problems.

### Full version info

```shell
OS: macOS 14.7.6
Architecture: x86_64
Version: 1.1.0dev.42149 (Git)
Build date: 2025/06/13 08:44:01
Build type: Release
Branch: main
Hash: 9714e2f87a6240d1b2b979eb39c7d34791aafdae
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 72/127.75/2
Installed mods: 
  * fasteners 0.5.29
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

matthiasdanner, 2025-06-16:
MarkerSize was also changed with #21098. can you try with #21852?

maxwxyz, 2025-06-16:
@jacquesbeaurain FYI

matthiasdanner, 2025-06-16:
@garyrussell @maxwxyz seems to work with #21852 again do you mind to test it yourself?

[image]

[image]

JohnOCFII, 2025-06-16:
There is definitely an issue with large marker sizes in the current code base.  Now, if @garyrussell happens to have a Retina screen on his Mac, he may find he no longer needs such a larger marker, due to the other changes to increase the size on higher resolution screens.

When I choose a marker size of 25px, the markers seem to disappear:

[image]

At a marker size of 9px, it looks correct to me:

[image]

This is on a 2021 M1 MacBook Pro with 16" display.

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Unknown
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.6.1
  * Telemetry 1.0.1
```

garyrussell, 2025-06-16:
@JohnOCFII thanks!! I never thought of trying to reduce the size.

I have an older MacBook Air with a Retina display.

Anything up to 15px works; 20 and above causes them to disappear.

Since there is a workaround, this issue can have a lower priority from my perspective. But I gather this might be one symptom of a larger problem.

Thanks for the quick response.

matthiasdanner, 2025-06-17:
@maxwxyz the problem here is also the scaling like in #21745 

The markers are bitmaps and they are created in MarkerBitmaps.cpp to a certain size. And if the size is too big the bitmap for 7px is drawn (see MarkerBitmaps::getMarkerIndex).
Because of the scaling you get a bitmap a few sizes bigger than what you have chosen as marker size. 
So if you choose 7 you get the bitmap for 13px 
if you choose 15 you get the one for 30 
and if you choose 20 or more as there is no bigger bitmap you get the one for 7 px.

so I think this is fixed if the scaling is fixed correctly.
you could add more sizes for the markers but I think that's not the correct way to fix this

paranoidi, 2025-11-29:
@luzpaz 

Actually let me re-raise this issue which was closed without addressing the problem.

https://github.com/FreeCAD/FreeCAD/issues/24151

Whole vertex selection is still pretty much unusable (at least in hidpi monitors).

Here is a video with 1.1 RC1 using largest working marker size (15px). 

https://github.com/user-attachments/assets/ee8111ec-8741-423b-965f-50acaddd22ef

As you can see most of the area within vertex is not considered as highlighted / selected.