FreeCAD/FreeCAD #22030: Two subtractive helixes generate artifacts

h3ct0r, 2025-06-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I'm trying to generate a self reversing screw using two subtractive helixes, but it seems that this particular set of steps is triggering errors in the model with multiple `Refine failed: Removing splitter failed` messages and the model with artifacts. Messing around with the height parameter of the failing helix sometimes improve the model, but its not predictable what or when it will break.

Disabling the **refine checkbox** in the Helix panel did not improve the situation. 

Link to sample file: https://www.dropbox.com/scl/fi/jmzh90ghnhma2qt8t8jqf/self_reversing_test.FCStd

[image]

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * freecad.gears 1.3.0
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Anything else?

[self_reversing_test.FCStd.zip](https://github.com/user-attachments/files/20767784/self_reversing_test.FCStd.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Uthedris, 2025-08-19:
This can also happen with just a single subtractive helix.  

This is what the cut should look like:

[image]

But if I ask for one fewer turn I get this:

[image]

h3ct0r, 2025-10-05:
Sure, this is the zipped file:

[self_reversing_test.FCStd.zip](https://github.com/user-attachments/files/22711158/self_reversing_test.FCStd.zip)