FreeCAD/FreeCAD #22031: <PartDesign> FeatureThickness.cpp(152): Exception on making thick solid: BRep_API: command not done Thickness: Failed to make thick solid

Harvie, 2025-06-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I cannot apply thickness to my part. It's relatively simple geometry, but defined by compound curves:

[image]

Gives me
```
<PartDesign> FeatureThickness.cpp(152): Exception on making thick solid: BRep_API: command not done Thickness: Failed to make thick solid
```

I am trying to apply thickness on bottom face of Body001 - Fillet: [octavia3-dash-plate.FCStd.zip](https://github.com/user-attachments/files/20768626/octavia3-dash-plate.FCStd.zip)

### Full version info

```shell
OS: Arch Linux (XFCE/Xfce Session/xcb)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Czech/Czechia (cs_CZ)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Turntable/Object center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 90/172.811/1
Installed mods: 
  * sheetmetal 0.7.22
  * Curves 0.6.58
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-08-05:
I can confirm the issue. I have been trying to make this tutorial (https://www.youtube.com/watch?v=HnOTIZq72Yc ). He is using 0.21 version  and at 4:10  he gives thickness to object but when I do the same thing FC gives error messages  

```
<PartDesign> FeatureThickness.cpp(152): Exception on making thick solid: BRep_API: command not done
10:48:33  Thickness: Failed to make thick solid
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42837 (Git)
Build date: 2025/08/04 17:39:07
Build type: Release
Branch: main
Hash: 7d6c9c0ee601eb62717271f7cca9e6c1b34432a1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61
```

[thickness_test.zip](https://github.com/user-attachments/files/21593932/thickness_test.zip)

https://github.com/user-attachments/assets/cfc50eb2-95c4-424a-887c-626c3b9562fb

MikePryadko, 2025-10-07:
**UPD** Strange solution found!

https://github.com/user-attachments/assets/4644f89a-d282-44a9-af2e-a3d0435eb476

On this video:
1) open my example project (`4_by_120_smooth.FCStd`)
2) try to make `Thickness` and get error
3) remove `Fillet`
4) **go to sketch's property and rotate to `45*`**
5) create new `Fillet`
6) create `Thickness`
7) get SUCCESS! 😁 

Such a crazy magic))

Harvie, 2026-01-28:
Do we need to file OCC issue somewhere upstream?

luzpaz, 2026-09-12:
@pieterhijma is there an upstream ticket for this yet ?