FreeCAD/FreeCAD #22032: doesn't work

GorillaSapiens, 2025-06-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

i'm a new user.  this is a flaming dumpster heap of trash.  i have an STL file.  i just want to do some measurements.  20 minutes of bungling about to make it a mesh then an object?  the whole "workbench" thing is confusing.  tools/measure is greyed out now, but when it did do something it ignored where i was clicking and just drew a line off into the corner.  very frustrated.  uninstalling.  not a tool for new users.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-06-17:
You can open a topic with images and file on https://forum.freecad.org/

Zaubentrucker, 2025-06-17:
Feel free to get a refund for all the money that you spent on this CAD suite

xtemp09, 2025-06-17:
@GorillaSapiens, just watch educational videos on _YouTube_ and read some wiki before doing something. You are right about one thing ― this is not for newcomers. Perhaps, _Google SketchUp_ or _MeshLab_ will suit your needs.