FreeCAD/FreeCAD #22034: Sketcher: Constraints have 'First' 'Second' 'Third' which is limiting

PaddleStroke, 2025-06-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Constraints currently have 

```
    int First;
    PointPos FirstPos;
    int Second;
    PointPos SecondPos;
    int Third;
    PointPos ThirdPos;
```

Which is very limiting because we can't have more than that
I'm working on a 'group' constraint and I need unlimited number of geo elements

So I am planning to replace those by a std::vector<GeoElements>

This sounds like it's going to be a massive change.

Headsup @FlachyJoe @AjinkyaDahale @chennes 

### Full version info

```shell
1.1
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-06-17:
Replacing First ... might makes it difficult to support backward compatibility...
Alternatively I add a `std::vector<GeoElementId>` on top and only use it for the constraints that needs it.

FlachyJoe, 2025-06-17:
I'm ok too.

> might makes it impossible to support backward compatibility

It needs a further look but I think the ids are only saved as integer so it wouldn't be a problem to map them back to the vector content.
If in the FCStd file the data look like
`Constraint Distance 5 12 2.5`
with `5 12` the geoId it acts on, there is no problem with internal usage.

AjinkyaDahale, 2025-06-18:
@PaddleStroke I'd like to know how you plan to handle things on the gcs solver front. There, every constraint reduces _exactly one_ degree of freedom.

FlachyJoe, 2025-06-18:
@AjinkyaDahale I think Paddle plane a Sketcher complex constraint reduced to several basic GCS constraints. If it's not the case I agree it will certainly cause damage…

hyarion, 2025-06-18:
@PaddleStroke can you explain how group constraints would work and why you need it?