FreeCAD/FreeCAD #22042: Expression Editor do not list BoundBox properties

impeeza, 2025-06-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Hi, first at all, thanks a lot for your great work and all effort put on the FreeCAD.

I recently start to note what if you are using Expression Editor to reference a formula to values on other element, and try to use BoundBox the autocomplete assistant get lost.

First, BoundBox is not displayed as a possible selection

[image]

[image]

But manually writing «BoundBox» on the text box the Expression is evaluated and the value of the BoundBox is displayed.

[image]

But then putting a «.» after the word BoundBox, do not show the properties and the expression become invalid
[image]

[image]

Manually writting a property like «XLength» the value is correctly evaluated.
[image]

### Full version info

```shell
OS: Windows 11 build 22621
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * ThreadProfile 1.96.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

tetektoza, 2025-10-03:
@impeeza haha, this made me randomly happy. Let me know if everything works! I can always modify it :)

impeeza, 2025-10-03:
Mate,  I am glad you liked, anybody had to be recognized their hard work, regards.