FreeCAD/FreeCAD #22053: ToolName subitituions not generated when split by operation

firebladed, 2025-06-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When Post Processing Output from a CAM job the tool controller names (%t) and number (%T) are not being included when OrderOutputBy is set to Operation

e.g when PostProcessOutputFile is set to "%d_%j-%S_%t_%T_%O" 

i get
Unnamed_Job-0___-3mm Drilling.nc
Unnamed_Job-1___-6mm Drilling.nc
Unnamed_Job-2___-Adaptive.nc
Unnamed_Job-3___-Profile.nc

Unless there is a multiple tool operation i don't know about the tool for each operation should always be known

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly4 0.50.18
  * Assembly4.backup1714492159.8689911 (Disabled)
  * Assembly4.backup1749489350.702833 0.50.18 (Disabled)
  * boltsfc 2022.11.5
  * btl 0.9.9
  * EM 2.1.1
  * fasteners 0.5.38
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.1
  * OpticsWorkbench.backup1714492175.9109938 (Disabled)

---------

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42095 (Git)
Build date: 2025/06/06 16:36:48
Build type: Release
Branch: main
Hash: 9c017b0dbf538c2570df7c3316355cdcdc3163f5
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * 3D_Printing_Tools
  * A2plus 0.4.68
  * Assembly4 0.50.18
  * Assembly4.backup1714492159.8689911 (Disabled)
  * Assembly4.backup1749489350.702833 0.50.18 (Disabled)
  * boltsfc 2022.11.5
  * btl 0.9.9
  * EM 2.1.1
  * fasteners 0.5.38
  * Help 1.0.3
  * kicadStepUpMod 11.3.9
  * Manipulator 1.6.0
  * OpticsWorkbench 1.2.1
  * OpticsWorkbench.backup1714492175.9109938 (Disabled)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

firebladed, 2025-06-18:
Example Document with multiple operations with different tools sorted by operation

[Unnamed.zip](https://github.com/user-attachments/files/20801834/Unnamed.zip)

Connor9220, 2025-09-05:
I think this might be caused by the fact that the name tool name / number can't be found.  Could be something related to the new tool library stuff.

sliptonic, 2025-09-06:
Actually,   From what I can see, this is working correctly.  

The tooltip for substitution strings is misleading though.

If splitting is **disabled**, all of these are ignored.
    %S ... Sequence Number (default)
    %T ... Tool Number
    %t ... Tool Controller label
    %W ... Work Coordinate System
    %O ... Operation Label

This makes sense because the combined file will potentially contain multiple of each so it's not possible to include them.
The internal logic is such that it ignores the unused substitutions.

If splitting is **enabled**, then the substitution string can include information to make the split filenames meaningful but it will ignore other information.  

|split on| use | Ignore |
|-----------|-------|--------|
|fixture | %W | %O %T %t |
|Operation| %O | %T %t %W |
|Tool| **Either %T or %t** | %O %W |

The confusing bit is that for split on tool,  it will use EITHER the tool number or the tool label, whichever is first.  And for split on operation, where including the tool should be possible, it ignores it altogether.

So, In your job you have it set to split on operation and you're getting:

Unnamed_Job-0___-3mm Drilling.nc
Unnamed_Job-1___-6mm Drilling.nc
Unnamed_Job-2___-Adaptive.nc
Unnamed_Job-3___-Profile.nc

It's a bit confusing that you have two operations that are named the same as tool controllers but all the names are definitely coming from the operations.

What looks like missing data in ___ is the ignored substitutions

Bottom line, this is working as designed and there's no bug.  It can be improved, of course, to include %T and %t when possible but that would be a new feature.  I'm removing the bug label but leaving this issue open.  Perhaps we can revisit it after a release.

closed 2025-10-16 by sliptonic