FreeCAD/FreeCAD #22069: Draft: Draft_Hatch enable creation of multiple hatches at once

semhustej, 2025-06-19:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

**Feature request**

**Current situation**
When user wants to create hatch pattern on multiple faces, he/she has to create hatch for each face independently. This can be quite tedious in case like hatching walls plan view, where there can be a lot of independent hatch objects.

See current behavior in the video:

https://github.com/user-attachments/assets/ea2c3155-e38a-4b7c-a0cb-c825201ec42d

**Possible solution**
Enable creation of multiple hatches with one command, when at the beginning of the command several faces are selected.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

yorikvanhavre, 2025-07-12:
I implemented that in the above PR. Please test!

ErtugrulEge, 2025-10-04:
I tested  but I didn't see anything changing. 

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.61

```

semhustej, 2025-10-04:
> I tested but I didn't see anything changing.
> 

It is not in the weeklies yet. The PR is only a draft.