FreeCAD/FreeCAD #22076: Adaptive: 'wire is not closed' when a intersecting holes are added to part.

RiccaDS, 2025-06-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Adaptive works fine on this hole: 

[image]
but if I set the tip to the next(and last) feature, the preceding adaptive operation no longer works throwing a "wire is not closed" error. The part doesn't seem to be broken, but something messes up with adaptive: 

[image]

### Full version info

```shell
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * Curves 0.6.63
  * Manipulator 1.6.0
  * Estimate 0.1.2
  * fasteners 0.5.38
  * ThreadProfile 1.96.0
  * Cables 0.2.0
  * Defeaturing 1.2.6
  * A2plus 0.4.68
  * FeedsAndSpeeds 0.6.0
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dbtayl, 2025-06-25:
Very likely a duplicate of https://github.com/FreeCAD/FreeCAD/issues/22073.

Fix is stuck on finding a solution to cleanly projecting arcs/splines to the XY plane: https://forum.freecad.org/viewtopic.php?t=96716

FEA-eng, 2026-08-29:
> This particular example isn't triggering any issues for me now. Possible it was fixed by [6c66564](https://github.com/FreeCAD/FreeCAD/commit/6c665640620d38730c2d323606b391998beec919)
> 
> There may well be other issues, but this one at least seems fixed to me.

@maxwxyz I guess it can be closed then.

closed 2026-08-29 by maxwxyz