FreeCAD/FreeCAD #22078: Error when trying to export as DXF: cannot access local variable 'tess'

garywill, 2025-06-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Trying to export as DXF.
If I check using legacy dxf importer or exporter, will get error:
```
pyException: Traceback (most recent call last):
  File "<string>", line 8, in <module>
  File "/tmp/.mount_FreeCAeeJjIm/usr/Mod/Draft/importDWG.py", line 129, in export
    importDXF.export(objectslist, dxf)
  File "/tmp/.mount_FreeCAeeJjIm/usr/Mod/Draft/importDXF.py", line 3798, in export
    sh = projectShape(ob.Shape, direction, tess)
                                ^^^^^^^^^
<class 'UnboundLocalError'>: cannot access local variable 'tess' where it is not associated with a value
```
I had to use legacy importer/exporter if I need to export the 3D model's current view to 2D. The new exporter always export top view. (Also we hope improvement of new exporter)

### Full version info

```shell
OS: openSUSE Leap 15.6 (KDE/plasma5/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

Could successfully export on FreeCAD 1.0.0 with legacy exporter.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

garywill, 2025-06-21:
> Can you share the file with which you're getting the error?

Create any new project with a simple object will reproduce the bug. I've tried. 
(Just create a box or any simple object on Part)

Also, I found another guy reporting same bug here https://forum.freecad.org/viewtopic.php?t=97428

semhustej, 2025-06-21:
I can reproduce the issue in FreeCAD 1.0.1

I am not able to reproduce with FC 1.1dev version.

@garywill can you please try to test with the latest weekly build? You can download it from here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1

```

garywill, 2025-06-21:
Tested with `FreeCAD_weekly-builds-42180-conda-Linux-x86_64-py311.AppImage`. No problem

closed 2025-06-23 by maxwxyz