FreeCAD/FreeCAD #22099: Reproducible crash when deleting item from tree

dagbdagb, 2025-06-20:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Whenever I delete the last item in my tree, freecad crashes. This with a fresh checkout/build from git tonight.
No clue how that item got into the tree, but freecad does not consider the document broken/inconsistent. And I cannot delete it without a crash.

[image]

I notice I am not allowed to upload my .FCStd-file,  What is the common practice for this in the freecad world?

Log with trace and more  is attached.
[mylog.txt](https://github.com/user-attachments/files/20842392/mylog.txt)

I am running with:
COIN_GL_NO_CURRENT_CONTEXT_CHECK=1

Let me know what else I can offer to hunt this one down.

### Full version info

```shell
OS: Gentoo Linux (Wayfire/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.42203 (Git)
Build date: 2025/06/20 17:21:14
Build type: Release
Branch: main
Hash: 889d708992d912f5f33d1cc01d11173495f16e16
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/129.483/1
Installed mods: 
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-21:
You can upload a .zip of your file.
Is it just with measurement objects or any object which is the last?

dagbdagb, 2025-06-21:
> You can upload a .zip of your file. Is it just with measurement objects or any object which is the last?

I started to happen with that exact object, in that exact file. 
But I started a new file right now, and yes, freecad crashes the moment I make another measurement. Steps:

Part Design
File | New
Body
Create Sketch
20 x 20 square
Pad 10
add measurement using origo and the corner at 0,0,20 as points.

Prior to this I had been playing with LC_* vars to quieten the 'posix collate' warnings which bothered me. Unsetting these variables makes no difference, so likely a red herring.

ErtugrulEge, 2025-06-21:
When I press delete button on your file I get access violation. Then I made it from scratch I couldn't delete measurement also you can get an error when you use measurement->position option.

**That seems to me  the problem is about  measurement->position also center of mass is not worked for me.**

[image]

[image]

[image]

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
Installed mods: 
  * Curves 0.6.61
```

dagbdagb, 2025-06-24:
My issue appears resolved now. Closing ticket.

ErtugrulEge, 2025-06-24:
I don't think so

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

[image]

dagbdagb, 2025-06-25:
OK, something is still amiss. Freecad no longer crashes. But there are error messages in the log. Will update this ticket later with exact messages,

dagbdagb, 2025-06-25:
Using the same binary as yesterday. 
Deleting a measurement from a previously saved document works without freecad crashing.

Trying to add a measurement now yields:

<Exception> 'App::FeaturePythonT<FeatureT>' is not a document object type
1.22e-07 Selection.cpp(158): Unhandled Base::Exception caught in selection observer: 

No selection is added.

I can also provoke the following messages:

No measure handler available for geometry of module: App
No measure handler available for geometry of module: Measure

This *may* be due to  the feature being confusing. Or myself being confused. Or a combination thereof.

Syres916, 2025-06-25:
Using the `PartDesignExample.FCStd` shipped with FreeCAD, other than `CenterOfMass` which seems to have many bugs, I can create measurements, delete measurements, save as a new filename and reopen and start adding new measurements without a single error in the Report View using:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42338 (Git)
Build date: 2025/06/25 19:40:05
Build type: Release
Branch: main
Hash: 8fe791bd6f11bd22563963035e77012fce41a553
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.917/1
Installed mods: 
  * MeshRemodel 1.10.37
  * FreeCAD-themes 2025.6.4
  * Assembly4 0.50.18
  * CfdOF 1.30.4
  * Dracula 0.0.6
  * Woodworking 1.0.1.39285.1.0.1.2025.06.04
  * sheetmetal 0.7.23
  * fasteners 0.5.38
```
So without a file and a exact procedure using at least the same Git Revision as above I'm struggling to replicate the `Selection.cpp` error above. One thing that I find annoying is how many times the measurements are left to be `Recompute`'d even though the model itself has never changed and none of the Sketches have even been opened.

dagbdagb, 2025-06-27:
The file attached to this ticket [crash.FCStd.zip](https://github.com/user-attachments/files/20846024/crash.FCStd.zip) has an attachment I cannot delete. Console states "No measure handler available for geometry of module: Measure"

This measurement (in the attached file) may appear a bit odd. I have another file where my measurement is a Length with a value. I get the same console response there, and I cannot delete this measurement either.

dagbdagb, 2025-06-27:
But: freecad is not crashing. Currently at:

OS: Gentoo Linux (Wayfire/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.13.5, Qt 6.9.1, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.1
shiboken 6.9.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: C/Default (C)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Classic/
Logical DPI/Physical DPI/Pixel Ratio: 96/129.483/1
Installed mods: 
  * fasteners 0.5.38

maxwxyz, 2025-06-28:
So it is not crashing on latest main?

ErtugrulEge, 2025-06-28:
> I don't think so
> 
> OS: Windows 10 build 19045 Architecture: x86_64 Version: 1.1.0dev.42180 (Git) Build date: 2025/06/16 16:25:59 Build type: Release Branch: main Hash: [fca41e4](https://github.com/FreeCAD/FreeCAD/commit/fca41e4003f1680cf604b593a046ac2b1db6ba8a) Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3 shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1 Locale: Turkish/Türkiye (tr_TR) Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1 Installed mods:
> 
>     * Curves 0.6.61
> 
>     * fasteners 0.5.38
> 
> 
> [image]

Now, I can delete the position dimensions with latest weekly. No crash or no error messages 

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

maxwxyz, 2025-06-28:
thanks, closing then.

closed 2025-06-28 by maxwxyz