FreeCAD/FreeCAD #22101: TechDraw: Display issue with B-Splines with certain scale. (1.1.0dev)

wiltur, 2025-06-21:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

When you insert a view of an object containing B-Splines, the view in TechDraw may not correspond to the sketch at all.
This seems to depend on the scale you apply to the view, in the example I've attached, the 1:1 and 
1:2 scale views are distorted, the only one that corresponds to reality is the 2:1 view.

This has happened to me on more complex projects where only the 1:5 scale gives a correct result.

The sketch used for the object is entirely constrained.

[Scale-issue.pdf](https://github.com/user-attachments/files/20844890/Scale-issue.pdf)

[image]

[image]

### Full version info

```shell
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git) AppImage
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: French/France (fr_FR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/87.0372/1.25
Installed mods: 
  * Curves 0.6.63
  * parts_library.backup1738827813.9563386
  * parts_library
  * fasteners 0.5.38
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ryankembrey, 2025-06-21:
I have reproduced this error.

[image]

File if anyone else wants to play with it.

[b-spline-test.zip](https://github.com/user-attachments/files/20846475/b-spline-test.zip)

WandererFan, 2025-07-17:
This problem is in the interpretation of bsplines as circles.  There is an error tolerance in the decision making code, if the error is less than the tolerance, we pretend the spline is a circle.

I can fix this easily, by changing that tolerance, but I don't know what that would do to existing drawings.  It would prevent some splines that would currently be treated as circles from being dimensioned as circles.

Some input from subject matter experts would be helpful here.

[image]