FreeCAD/FreeCAD #22125: BIM: Arch_Pipe with Arch_Connector deletion causes error

semhustej, 2025-06-22:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

- open the file (remove the .zip extension)

[FC-pipe-connector-delete.FCStd.zip](https://github.com/user-attachments/files/20851924/FC-pipe-connector-delete.FCStd.zip)

- select `Pipe`, `Pipe001` and `Connector` objects
- delete these objects
- error shows up
```
pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'OSError'>: Access violation
```

- the `Connector` object is not deleted

See behavior in the video:

https://github.com/user-attachments/assets/1363ede1-88d1-4a77-a398-ef3223d577d9

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * Reporting
  * Telemetry 1.0.1
```

### Subproject(s) affected?

BIM

### Anything else?

I am not able to reproduce the behavior in FC 1.0.1. Seems like a regression.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-06-22:
Can be related with https://github.com/FreeCAD/FreeCAD/issues/21653  and https://github.com/FreeCAD/FreeCAD/issues/22099#issuecomment-2993634335

Syres916, 2025-06-25:
This is actually fixed while I understand that doesn't help with the fact that some of the Weekly builds aren't updating. Tested using:

```
OS: EndeavourOS (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.42322 (Git)
Build date: 2025/06/24 08:45:46
Build type: Release
Branch: main
Hash: 27c786fef10cae96fc8f6e51ac617abd59ae8d45
Python 3.13.3, Qt 6.9.0, Coin 4.0.3, Vtk 9.4.2, boost 1_88, Eigen3 3.4.0, PySide 6.9.0
shiboken 6.9.0, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: Dracula.qss/Behave-dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.917/1
Installed mods: 
  * MeshRemodel 1.10.37
  * FreeCAD-themes 2025.6.4
  * Assembly4 0.50.18
  * CfdOF 1.30.4
  * Dracula 0.0.6
  * Woodworking 1.0.1.39285.1.0.1.2025.06.04
  * sheetmetal 0.7.22
  * fasteners 0.5.38
```

semhustej, 2025-06-27:
Tested with latest 1.1dev release and it seems to be fixed. Thank you.

I am closing the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1

```

closed 2025-06-27 by semhustej