FreeCAD/FreeCAD #22142: Part Design: Additive helix based on a Shapebinder is yellow and transparent

FEA-eng, 2025-06-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

As mentioned [here](https://github.com/FreeCAD/FreeCAD/pull/12207#issuecomment-2993807842), Additive helix based on a Shapebinder (the blue one) is yellow and transparent:

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-06-26:
cc @FlachyJoe per https://github.com/FreeCAD/FreeCAD/pull/12207#issuecomment-2993807842

FlachyJoe, 2025-06-26:
It's not specific to hell-x but all additive features.
And it's only the case if the ShapeBinder is alone in its Body. If a standard shape is present then the generated shape looks normal.
SubShapeBinder is not affected.
[image]

FlachyJoe, 2025-06-26:
@luzpaz <strike>helix</strike> → feature ?

maxwxyz, 2025-06-30:
@kadet1090 FYI crash on testing transparent previews

luzpaz, 2025-07-27:
cc @kadet1090 👆

kadet1090, 2025-08-12:
@FEA-eng can you test with current main?

FEA-eng, 2025-08-13:
> [@FEA-eng](https://github.com/FEA-eng) can you test with current main?

It's still the same there.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```