FreeCAD/FreeCAD #22145: Creating windows and doors for rooms smaller than 1.5mx1.5m creates holes through everything.

lboclboc, 2025-06-23:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The title actually says it all. I created a room using sketcher with wall thickness of 100mm left aligned. When creating a door or window, the hole created by the object will go through the whole host wall.
Also its not possible to delete the window or door, or modify its properties after its created.
In the log window I see:
14:41:11  Part::FeaturePython: Link(s) to object(s) 'Wall' go out of the allowed scope 'Window'. Instead, the linked object(s) reside within 'Part'.

When trying to delete the door I get the log message:
14:42:09  378.272 Selection.cpp(1518): Sub-object Nytt_Badrum#Part.Wall.Window. not found

### Full version info

```shell
OS: KDE Flatpak runtime (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.40267 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 04d2bcb)
Hash: 04d2bcbf45ab96f3895ac3d0e8b5ae9becbaf5cd
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical/physical DPI: 96/175.939

Attached my file with a broken door.
```

### Subproject(s) affected?

None

### Anything else?

[Bad-Bathroom.FCStd.gz](https://github.com/user-attachments/files/20865401/Bad-Bathroom.FCStd.gz)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

lboclboc, 2025-06-23:
Sorry, I can't really find the "Bim.Door.Hole_Depth" settings anywhere in my version.

paullee0, 2025-06-23:
OK, I'll have a a look when there is a gap, thanks.

lboclboc, 2025-06-23:
I see what is going on now, when I create a big enough room and create a door I see a whole depth property for that door. The problem is that when I create the door for the small room, I can not change properties for the door because its in some kind of error state.

lboclboc, 2025-06-23:
A workaround seems to be to create a standalone door, set the hole dephth for that and finally change its host.

paullee0, 2025-06-23:
In dev weekly build like git 41736, just recompute the Wall and it should be fine:

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.41736 (Git)
Build date: 2025/05/19 10:15:51
Build type: Release
Branch: main
Hash: 76ce8ccfd5b52e520c052d2193d7a244e83d4cb6
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/30.48/1
Installed mods: 
  * Link to SketchArchPlus
```

[image]

closed 2025-06-23 by maxwxyz