FreeCAD/FreeCAD #22161: TechDraw: Views added to a clipgroup can become unfindable

dannystaple, 2025-06-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Reproducing the problem:

1. Create a new file in FreeCAD.
2. Create a body.
3. Sketch something - ie a rectangle with a corner cut out will do, but any shape really, boring I know, in part design, dimension the sketch to constrain sensibly (I went for 10mm, by 10mm, with a 3 by 4 cutout).
4. Pad 10mm.
5. Go to tech draw, insert a page (I use a4 landscape blank usually)
6. Select the body from the object tree, insert view. 
7. I selected a few views - front, top, left, topleft (it might be reproducible with fewer) - select Ok.
8. Ensure frames are visible
9. Select insert clip group
10. Resize to 5mm x 5mm (or make a larger shape see above, clip group should be smaller than the view)
11. Add a view, e.g. top into the clipgroup:
    - In properties, find the views attribute, select the top view.

- The top view is in the property panel as both a child of the clip group and of the views ProjGroup

## Expected result:

- I can see the clipgroup in the page view
- I can see the top view and position it within the clip group in the page view

## Actual result

- The clipgroup is visibly empty in the page view
- The top view is totally hidden from the page view

## Workaround

- I tried adjusting the top view x/y coordinates in the copy under the clip group - doesn't work.
- Working (ish) - make the clip group larger than the view, move the projection view into position. The position needs to be relative to the *middle* of the clip group to resize smaller again.
  - Further observation - any update to the objects, clip group or views, can lead to the relative position of the view inside the clipgroup to be reset. Disabling auto-distribute on the ProjGroup seems to help here (and might be a useful tip in clipgroup documentation).
  - It seems to be locked in one axis - ie the top view can move vertically, but not horizontally, this might be due to it's projgroup relationships more than the clip group.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * ExplodedAssembly
  * fasteners 0.5.38
  * kicadStepUpMod 11.6.9
  * Manipulator 1.6.0
  * Telemetry 1.0.1
```

### Subproject(s) affected?

TechDraw

### Anything else?

[clip-group-example.zip](https://github.com/user-attachments/files/20882979/clip-group-example.zip)

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

dannystaple, 2025-06-24:
I was able to reproduce this by making a Projection view with only 1 view (not a group), so it stands alone.
It is now movable in X and Y, but still becomes hidden. 
The workaround - creating a bigger clip, getting the object positioned in the center and then narrowing, appears to work better with a view not connected to a projection group.

dannystaple, 2025-06-24:
For reference, what I am aiming at, and was able to get with the workaround:

[image]

dannystaple, 2025-06-24:
Trying this as a workflow for multiple clip groups in a more complicated project (where I have an assembly, and want to focus on one area of it from different views):
- Create the projection group.
- Ungroup it (no locking/auto positioning)-  by going to the "views" property in the data tab for the ProjGroup and deselecting the views. (note - these can end up above the page in the tree - slightly awkward).
- Create the first ClipGroup
- Add the view to it (filter by type and sleect techdraw views is handy in a large object tree)
- Set that views x,y to 0,.0 to center it (the frame isn't visible as it is larger than the clip group)
- Resize the clipgroup large enough to position the view (or use the properties panel to fine bump, you have to click away to let it recalculate when you do these changes)
- Resize the clip group down again
- Repeat for the oher views

This workflow still has some glitches:
- The anchor view ends up locked, and the lock property cannot be overridden back to unlocked.

dannystaple, 2025-06-24:
This might explain the anchor bit - https://github.com/FreeCAD/FreeCAD/blob/27c786fef10cae96fc8f6e51ac617abd59ae8d45/src/Mod/TechDraw/App/DrawProjGroupItem.cpp#L102 - if ungrouping does not clear the IsAnchor property, locked will be true.

WandererFan, 2025-06-27:
Views in a clipgroup have been hard to find if they are outside the clip window since clipgroups were invented.

Changing the projection group's Views property to move secondary views out of the projection group was not something considered in the original design, I think.

To unlock the Front view, send it to the python console and then run "obj.LockPosition = False".

dannystaple, 2025-06-29:
This was kinda the problem, when the clipgroup was created, the view was outside the clipgroup, and therefore couldn't be dragged, along with being locked. I was able to solve it in the python console by unlocking, and then see the view in the clipgroup by updating it's coordinates.

luzpaz, 2025-08-21:
Should i mark as 'confirmed' ?