FreeCAD/FreeCAD #22166: Transform tool is not working

zbynekwinkler, 2025-06-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Transform tool is not working. It was working in the previous weekly build from June 16th. Now nothing happens when Transform is selected.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/06/23 18:46:33
Build type: Release
Branch: main
Hash: 6faa377fbd6b7f8273f50c21a18fc004674325f9
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * Defeaturing 1.2.6
  * fasteners 0.5.38
```

### Subproject(s) affected?

None

### Anything else?

Steps to reproduce:
1. New file
2. Part workbench
3. Create cube (Part_Box)
4. from context menu of the created cube select `Transform` 

Nothing happens.

The last weekly shows working Transform gizmo.

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-06-24:
I couldn't reproduce with latest wekly 42180

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
  * fasteners 0.5.38

[image]

zbynekwinkler, 2025-06-24:
When I try to activate the tool I get this output in `Report View` and in terminal:

```
00:43:04  <Exception> ExtensionContainer::getExtension: No extension of given type available
```

I managed to get a local build and run under debugger. This is the callstack when the exception is thrown:

```
libFreeCADApp.so!App::ExtensionContainer::getExtension(const App::ExtensionContainer * this, Base::Type t, bool derived, bool no_except) (/home/zbynek/freecad-ws/freecad-source/src/App/ExtensionContainer.cpp:122)
libFreeCADGui.so!App::ExtensionContainer::getExtensionByType<App::LinkBaseExtension>(const App::ExtensionContainer * this, bool no_except, bool derived) (/home/zbynek/freecad-ws/freecad-source/src/App/ExtensionContainer.h:172)
libFreeCADGui.so!Gui::ViewProviderDragger::getPlacementProperty(const Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:177)
libFreeCADGui.so!Gui::ViewProviderDragger::getObjectPlacement(const Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:396)
libFreeCADGui.so!Gui::ViewProviderDragger::updateDraggerPosition(Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:470)
libFreeCADGui.so!Gui::ViewProviderDragger::onChanged(Gui::ViewProviderDragger * this, const App::Property * property) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:104)
libFreeCADGui.so!Gui::ViewProviderGeometryObject::onChanged(Gui::ViewProviderGeometryObject * this, const App::Property * prop) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderGeometryObject.cpp:158)
PartGui.so!PartGui::ViewProviderPartExt::onChanged(PartGui::ViewProviderPartExt * this, const App::Property * prop) (/home/zbynek/freecad-ws/freecad-source/src/Mod/Part/Gui/ViewProviderExt.cpp:396)
libFreeCADApp.so!App::Property::hasSetValue(App::Property * this) (/home/zbynek/freecad-ws/freecad-source/src/App/Property.cpp:274)
libFreeCADGui.so!Gui::ViewProviderDragger::setTransformOrigin(Gui::ViewProviderDragger * this, const Base::Placement & placement) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:93)
libFreeCADGui.so!Gui::ViewProviderDragger::resetTransformOrigin(Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:98)
libFreeCADGui.so!Gui::TaskTransform::TaskTransform(Gui::TaskTransform * this, Gui::ViewProviderDragger * vp, Gui::SoTransformDragger * dragger, QWidget * parent, App::SubObjectPlacementProvider * subObjectPlacemenProvider, App::CenterOfMassProvider * centerOfMassProvider) (/home/zbynek/freecad-ws/freecad-source/src/Gui/TaskTransform.cpp:100)
libFreeCADGui.so!Gui::TaskTransformDialog::TaskTransformDialog(Gui::TaskTransformDialog * this, Gui::ViewProviderDragger * vp, Gui::SoTransformDragger * dragger) (/home/zbynek/freecad-ws/freecad-source/src/Gui/TaskTransform.cpp:767)
libFreeCADGui.so!Gui::ViewProviderDragger::getTransformDialog(Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:112)
libFreeCADGui.so!Gui::ViewProviderDragger::setEdit(Gui::ViewProviderDragger * this) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:208)
PartGui.so!PartGui::ViewProviderPrimitive::setEdit(PartGui::ViewProviderPrimitive * this, int ModNum) (/home/zbynek/freecad-ws/freecad-source/src/Mod/Part/Gui/ViewProviderPrimitive.cpp:73)
libFreeCADGui.so!Gui::ViewProvider::startEditing(Gui::ViewProvider * this, int ModNum) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProvider.cpp:141)
libFreeCADGui.so!Gui::ViewProviderDragger::startEditing(Gui::ViewProviderDragger * this, int mode) (/home/zbynek/freecad-ws/freecad-source/src/Gui/ViewProviderDragger.cpp:134)
libFreeCADGui.so!Gui::DocumentP::tryStartEditing(Gui::DocumentP * this, Gui::ViewProviderDocumentObject * svp, App::DocumentObject * sobj, int ModNum) (/home/zbynek/freecad-ws/freecad-source/src/Gui/Document.cpp:292)
libFreeCADGui.so!Gui::DocumentP::tryStartEditing(Gui::DocumentP * this, Gui::ViewProviderDocumentObject * vp, App::DocumentObject * obj, const char * subname, int ModNum) (/home/zbynek/freecad-ws/freecad-source/src/Gui/Document.cpp:285)
```

It happened when I tried to step over this line:

https://github.com/FreeCAD/FreeCAD/blob/f3980c8a6fd65f82d3e61d229d26ad0eb57265c1/src/Gui/TaskTransform.cpp#L100

kadet1090, 2025-06-25:
Yes, I have fix already - pushing it in a minute. One would assume that `getExtension` returns nullptr when extension is not available... ;)

PR #22173 (2025-06-25): Gui: Use no_except=true for getExtensionByType in ViewProviderDragger
`getExtensionByType` requires `true` as parameter to not throw if extension is not found, I did not know that :)

Fixes: #22166

## Issues

## Before and After Images
commit 69a60a1b (2025-06-25): Gui: Use no_except=true for getExtensionByType in ViewProviderDragger

Fixes: #22166
closed 2025-06-25 by hyarion