FreeCAD/FreeCAD #22170: Link Lighting is not working well when the choice is "Two Side"

leoheck, 2025-06-24:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Enum, nested with the Link Lighting, is messing up the expected result.

Doesn't matter the value I put on Lighting, it does not change the result if I keep the nested Enum as it is.

To obtain the expected result, I have to set Lighting to `Two side`, but I also have to remove the `One side` from the beginning of the Enum.

This enum here:
[image]

A better example was posted here:
https://github.com/FreeCAD/FreeCAD/issues/20776#issuecomment-3001575890

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42180 (Git)
Build date: 2025/06/16 16:25:59
Build type: Release
Branch: main
Hash: fca41e4003f1680cf604b593a046ac2b1db6ba8a
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
Installed mods: 
  * Cables 0.2.0
  * Assembly4 0.50.19-dev
  * sheetmetal 0.7.22
  * DynamicData 2.74.0
  * Curves 0.6.63
  * InventorLoader 1.5.1
  * diff-tools 0.1.0
  * ose-piping
  * OpenTheme 2025.5.20
  * fasteners 0.5.38
  * osh-autodoc-workbench 0.2.3
  * OpticsWorkbench 1.2.2
  * lattice2 1.0.0
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-06-25:
Why can I modify this Enum manually?

Is it possible to modify this Enum programmatically with Python somehow?

luzpaz, 2025-12-30:
> I have no idea and don't know what is this Lighting property nor what it's doing.

@leoheck where is this property sourced from ? Is it from an Addon ?

leoheck, 2025-12-30:
I though It was from Freecad. Today I will have some 
time to check this and to make a file.

FEA-eng, 2026-09-30:
Still relevant

[Link light.FCStd.zip](https://github.com/user-attachments/files/32842202/Link.light.FCStd.zip)

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```