FreeCAD/FreeCAD #22178: CAM: Inconsistent and inaccurate collision detection

sliptonic, 2025-06-25:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The CAM workbench doesn't provide consistent tools for detecting whether a toolpath colllides with the model.

Detecting whether a tool will gouge the model, collide with some part of the machine, clamps, vises, or other workholding is an expected part of modern CAM solutions

This issue is created to start discussion about what a collision detection system should include and how it might be integrated.

### Full version info

```shell
any
```

### Subproject(s) affected?

None

### Anything else?

This link isn't intended to suggestion a solution.  It's given here to explain what we're talking about
https://www.youtube.com/watch?v=NfefGUV5guU

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

RiccaDS, 2025-06-25:
Nice pick. I never used collision detection. however:
I agree on your first 4 points.
For maximum flexibility:
-No, the UI should not refuse to post process. It should strongly warn(with option to disable warnings), but not refuse. 
-Collision detection should be ON by default(or many users might not discover it's existence), but one should be able to disable it and/or disable it's warnings.
I agree on the rest.

Connor9220, 2025-06-26:
I modeled my ISO30 tool holders in conjunction with my toolbits so that I could see them in the simulator..  it gives me a quasi collision detection after the job is ran by inspecting.  I can see if the tool holder hit anything. Doesn't really do anything for other types of collision.  Here is a link to that discussion: https://github.com/FreeCAD/FreeCAD/issues/18690#issuecomment-2562036713

freckletonj, 2026-08-16:
I've built something for our internals that's relevant and could be duplicated:

1. custom Op/Worktree Item/TaskPanel that finds all jobs, and allows stitching them into arrays/one cohesive g-code. Eg Job1 is at G54 and surfaces the top of the stock within which 4 parts live. Job 2 is at G54 an represents a single part, so my CAMExport thing arrays it within. Job 3 is at G55 and involves flipping/refixturing and gets arrayed. All gets baked into a single nc file (with handy parks added for stock flipping), and fully simulated and collision checked.
2. Because this is all together, collision detection is easy:
  a. check intersection of g-code with tool-D-expanded bounding boxes of fixtures
  b. update a (sub)dexel data structure to show the results of real tools on a real stock simulation, and check shank collisions against uncut material. This is fast using `numba`. LLMs are fantastic at translating to fast numba.
  c. check intersections of analytical or tool geom against a mesh of the part to check for gouging

Happy to take questions if you're interested in any piece of that.