FreeCAD/FreeCAD #22182: My YZ plane is not working properly!

ctrleather, 2025-06-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Wanting to do something in Part WB, I chose the YZ plane, but after closing the sketch, intending to change its position, I noticed that in Data the rotation axes are all 0.58 degrees (x - 0.58, y - 0.58, z - 0.58), the angle was 120 degrees, the position elements (x, y, z) were all 0.
[image]
I proceeded to checks and tests and each time I encountered the same situation. For example, I opened the YZ plane and, without drawing anything, closed it and the same result. Note that the other planes had the data I was used to, namely the XY plane had an angle of 0 degrees, the rotation axes x - 0, y - 0, z - 1,
[image]
 and the position elements 0, and the XZ plane had an angle of 90 degrees, the rotation axes x - 1, y - 0, z - 0, and the position elements 0.
[image]
To remove any doubt, I restarted the computer and even uninstalled and reinstalled the program from the official website, but I encountered the same problem. Moreover, having an older computer, I saw that an older version of FreeCAD also had the same problem.
I tried to fix the problem manually, that is, I replaced the data in the rotation axes with the data that seemed correct to me, that is, I alternated the angle with 0 degrees and 90 degrees, the rotation axes x - 0, y - 1, z - 0, but when I started to "rotate" the sketch, it moved in a very strange way.
I even turned to Chat GPT, but to no avail.
Until one more thing, to avoid any problem, I thought of executing the sketches in the other planes (XY and XZ) and positioning them with the help of the elements in those planes where I need them. But the problem persists.
What is happening? Is there an issue that I don't know or don't understand? Am I making a mistake somewhere or does your program, which is very useful to me in my work, have a problem that I don't know how to solve?
Thank you!
Nicolae Prepeliță

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Romanian/Romania (ro_RO)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.12
  * Curves 0.6.50
  * lattice2 1.0.0
  * Movie 2025.1.4
  * Render 2024.12.15
  * sheetmetal 0.7.22
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-26:
https://stackoverflow.com/questions/832805/euler-angles-vs-quaternions-problems-caused-by-the-tension-between-internal-s

ctrleather, 2025-06-26:
Hello!
Thank you!
I read the material in the link and I think I understand what's going on.
I'm not good enough at aeronautics to understand quaternions, and I don't intend to understand them, I'm not interested, but I'm good at technical drawing, in high school even having specialized classes, often drawing the attention of the specialized teacher that I don't draw well on the board.
Now I did some tests and noticed that the sketches in the three planes can be modified without problems, in all directions of rotation if after selecting the sketch in the tree, you open the layout window (by clicking with the mouse arrow on the three dots on the right) and modify from the drop-down list in the Rotation section in "Euler Angles".
[image]
I'm glad that I understood and solved the problem, with your help!
I was really worried that I wouldn't be able to solve the problem.
I'm curious if AutoCAD also works with quaternions.
Only good!

kaiwas, 2025-06-27:
> Your alternative Euler angles lead to different local X and Y axes which would confuse the user.

It would be great to be able to switch the rotations from quaternions to Euler angles in the property.
You may say that this is complicated, but this is the approach used in blender. It is very convenient for the user.

[image]