FreeCAD/FreeCAD #22187: Techdraw: In Cosmetic Line the 'Style' field in Format property does not affect the displayed line style when using the Python API.

psttq, 2025-06-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Steps to reproduce:

1) Create a CosmeticLine using ```view.makeCosmeticLine```
2) Get the CosmeticLine via ``view.CosmeticEdges``
3) Create a new format like ``format = line.Format``
and change it to`` format["style"] = 0`` (or any other number)
4) Apply the new format ```line.Format = format```
5) Reopen the drawing to apply the changes

At the same time, other fields like weight, color, and visible work correctly and are updated if you change the line format with a double click. But the style does not change in the Format property if you change it using the GUI.

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * OpenTheme 2024.9.1
  * retain-wall-constructor
```

### Subproject(s) affected?

TechDraw

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

benj5378, 2025-06-26:
I can confirm this issue

luzpaz, 2025-06-26:
@benj5378 About info ? 🙏

luzpaz, 2025-07-01:
cc @WandererFan

luzpaz, 2025-07-28:
Any diagnosis here ?

WandererFan, 2025-07-29:
On the list, but no eta.

FEA-eng, 2026-05-24:
The macro posted above throws this error for me:

```
10:52:38  pyException: Traceback (most recent call last):
  File "C:/Users/labmi/AppData/Roaming/FreeCAD/v1-1/Macro/test.FCMacro", line 9, in <module>
    template.Template = FreeCAD.getResourceDir() + "Mod/TechDraw/Templates/A0_Landscape_blank.svg"
    ^^^^^^^^^^^^^^^^^
<class 'RuntimeError'>: Could not read the new template file
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260519 (Git shallow)
Build date: 2026/05/19 20:59:45
Build type: Release
Branch: (HEAD detached at 5a5d8d042)
Hash: 5a5d8d0425fdb440fb5dcb47f1c48422d07d3267
Python 3.11.15, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

closed 2026-10-12 by maxwxyz