FreeCAD/FreeCAD #22197: Toggling the visibility of multiple features of a Body in a cyclic manner.

leoheck, 2025-06-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

FC could allow toggling the visibility of 2 features (or more) inside a Body, for quick comparison between them.

This approach works with Bodies and Parts, but not with Features inside the Body. 

This thing could work this way: Since only one feature can be visible, it could toggle the visibility of the selected Features in a cyclic way. 

You select 2 or more features, and when you hit Space, if one of the selected features is visible, it will make the next selected feature visible. If none of them is visible, then it will make the first selected feature visible. And the visible feature could cycle as the user hits Space.

If just one feature is selected, then the visibility would work as it already does, toggling the visibility on/off.

### Full version info

```shell
OS: Ubuntu 25.04 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/94.0767/1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

leoheck, 2025-06-27:
Isn't this idea just a improvement of the current behavior of the toggle visibility function? 

When toggling visibility it should check the selection first to decide how the visibility has to be changed.

pierreporte, 2025-09-19:
I would not be against this feature.