FreeCAD/FreeCAD #22198: Syntax error on Sketcher workbench

acoustic-design, 2025-06-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

This is on a Mac
select sketcher workbench
Open create new document
Select create sketch
Choose orientation and click OK
error message "syntax error"

### Full version info

```shell
OS: macOS 15.5
Architecture: arm64
Version: 1.0.1.39285 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

luzpaz, 2025-07-21:
@acoustic-design can you test on 1.1-dev ? https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

FEA-eng, 2025-10-15:
@acoustic-design or @maxwxyz Can you check if it's still relevant on Mac?