FreeCAD/FreeCAD #22200: Transform Tool Not Working on Weekly 43325 On Windows

LHC-Canada, 2025-06-26:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

To Repeat
- open a new document in FreeCad
- Using Part Design, create simple sketch and then pad
- Select the part design body in the tree
- Right Click and select transform
Results
- nothing happens - transform dragger does not appear on screen
- Following error appears in report view:
- 14:04:48  <Exception> ExtensionContainer::getExtension: No extension of given type available

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/ProDark/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/112.94/1.25
Installed mods: 
  * A2plus 0.4.56a
  * Beltrami 1.1.0
  * CAM
  * Curves 0.4.4
  * Defeaturing
  * ExplodedAssembly
  * fasteners 0.3.44
  * fcgear 1.0.0
  * IconThemes
  * kicadStepUpMod 10.13.0
  * Manipulator 1.4.5
  * QuickMeasure 2022.10.28
  * scripts
  * sheetmetal 0.4.24
  * VendorParts
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

ErtugrulEge, 2025-06-26:
I can confirm. Can be rellated with https://github.com/FreeCAD/FreeCAD/issues/22166#event-18314234294  

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Turkish/Türkiye (tr_TR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/127.628/1
Installed mods: 
  * Curves 0.6.61
```
  * fasteners 0.5.38

[image]

Syres916, 2025-06-26:
It was fixed 3 commits after the 42325 build was created.