FreeCAD/FreeCAD #22205: Sketch Attachment Support Problem moving from 1.0 to 1.1dev

BackwardEcho, 2025-06-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I made a project in 1.0 and recently made the decision to update to 1.1dev. After opening the file I found all my sketches that were previously attached to a face to have broken their attachments. When I try to reset them, a face reference is not accepted by the Attachment Editor where previously I was able to, I can only select the body and then move sketch manually with coordinates.

[Attachment Problem.zip](https://github.com/user-attachments/files/20940788/Attachment.Problem.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/06/23 18:46:33
Build type: Release
Branch: main
Hash: 6faa377fbd6b7f8273f50c21a18fc004674325f9
Python 3.11.13, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/32.512/1.25
Installed mods: 
  * fasteners 0.5.38
  * FusedFilamentDesign 0.25.250
  * Telemetry 1.0.1
```

### Subproject(s) affected?

PartDesign

### Anything else?

I found [this issue](https://github.com/FreeCAD/FreeCAD/issues/18969) where sketches seem to have also lost their attachment when opening an older FC file in a new version of FC. Though theirs seem to be from 0.21 to 1.0 where mine is 1.0 to 1.1dev. I also don't know if he was able to reattach his sketches to face.

Hopefully the issue can be replicated as the "Attachment Problem" file was saved with 1.1dev and not 1.0. I can send the unedited version, but it has a lot of models in it. The file sent just has one.

[image]

[image]

[image]

[image]

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-06-27:
It's the location where you got the build from, I'm not sure why they are still being published as they are known to be faulty with regard to TopoNaming. Please try the Windows build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

BackwardEcho, 2025-06-27:
> It's the location where you got the build from, I'm not sure why they are still being published as they are known to be faulty with regard to TopoNaming. Please try the Windows build from https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Blender/Turntable/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/32.512/1.25
Installed mods: 
  * AddonManager 2025.6.1
  * fasteners 0.5.38
  * FusedFilamentDesign 0.25.250
  * Telemetry 1.0.1

That was it. I thought I asked what the difference was between the "-Bundle" URL and the version without, but I guess I didn't get an answer on that. Thank you.

For future reference if people check, https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.06.23 is the version I downloaded and had the issue with. The version Syres916 told me to try worked fine.