FreeCAD/FreeCAD #22207: BIM: BIM_ProjectManager shows warning

semhustej, 2025-06-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Priority: Low

When opening `BIM_ProjectManager` warning is shown. It has no effect on the functionality.

`Designer: The value "sizeAdjustPolicy" of the enum-type property QComboBox::AdjustToMinimumContentsLength could not be read.`

Steps to reproduce:

- open new file
- start `BIM_ProjectManager` tool
- warning shows up

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42325 (Git)
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.9477/1
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Syres916, 2025-06-27:
Just remove these three lines from the ui file and all will be fine https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/Resources/ui/dialogProjectManager.ui#L409-L411

semhustej, 2025-06-27:
> Just remove these three lines from the ui file and all will be fine https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/Resources/ui/dialogProjectManager.ui#L409-L411

OK, I'll do it. Thank you.

maxwxyz, 2025-06-27:
@semhustej did this fix it?
@Syres916 are you going to create a PR?

Syres916, 2025-06-28:
I was trying to help increase @semhustej commit count, if you're struggling with time, let me know.

semhustej, 2025-06-28:
> I was trying to help increase [@semhustej](https://github.com/semhustej) commit count, if you're struggling with time, let me know.

Thanks. I am not really hunting for commit counts :)

I have not fixed it yet. I will probably get to it in the next 14 days. But I am still not able to build FreeCAD myself, so it might be better for you to do the fix as I am not able to test the changes I do.

semhustej, 2025-06-30:
> 
> Done [#22243](https://github.com/FreeCAD/FreeCAD/pull/22243)

Thank you!