FreeCAD/FreeCAD #22216: Can't select Point

bjnobre, 2025-06-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

On dev version (last and previous build), I get this error bellow. The points and circle center become hidden when I move the mouse.

I've noticed if I create a file in dev version, even I open it in stable version the issue continues. If I create in stable version, issue don't occur, but when I open this stable file in dev version, issue appears only in dev version, unless I save the file in dev version. In this case, issue also occurs in stable version.

note: I've recorded screen with gnome built in screen recorder, so mouse isn't shown. 

https://github.com/user-attachments/assets/f48b7d95-9364-4ba9-910c-338824f64bac

### Full version info

```shell
OS: Debian GNU/Linux trixie/sid (GNOME/gnome/wayland)
Architecture: x86_64
Version: 1.1.0dev.42325 (Git) AppImage
Build date: 2025/06/24 17:49:05
Build type: Release
Branch: main
Hash: f3980c8a6fd65f82d3e61d229d26ad0eb57265c1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/143.907/1
Installed mods: 
  * Glass
  * fasteners 0.5.38
  * IconThemes

# System Details Report
---

## Report details
- **Date generated:**                              2025-06-27 13:25:47

## Hardware Information:
- **Hardware Model:**                              Dell Inc. Precision 3571
- **Memory:**                                      32.0 GiB
- **Processor:**                                   12th Gen Intel® Core™ i9-12900H × 20
- **Graphics:**                                    Intel® Graphics (ADL GT2)
- **Graphics 1:**                                  NVIDIA RTX A2000 8GB Laptop GPU
- **Disk Capacity:**                               1.0 TB

## Software Information:
- **Firmware Version:**                            1.30.1
- **OS Name:**                                     Debian GNU/Linux trixie/sid
- **OS Build:**                                    20241215T202556Z
- **OS Type:**                                     64-bit
- **GNOME Version:**                               48
- **Windowing System:**                            Wayland
- **Kernel Version:**                              Linux 6.12.32-amd64
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

FlachyJoe, 2025-06-27:
I have got a similar problem a few time ago but I pull the main git branch and it was solved so I didn't report.

@bjnobre Can you upgrade to the current dev and retry?

bjnobre, 2025-06-27:
I am actually on build 42325.

bjnobre, 2025-06-27:
I am not using from source code. I am using appimage weekly build. I'll wait for 42341 as appimage.

maxwxyz, 2025-06-28:
@bjnobre there is a new one here: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Syres916, 2025-06-28:
Previously reported [here](https://github.com/FreeCAD/FreeCAD/issues/21950) and point two in [Issue](https://github.com/FreeCAD/FreeCAD/issues/21915)

Syres916, 2025-06-29:
OK I was wrong with regard to the root cause of this issue, Git Bisect determined https://github.com/FreeCAD/FreeCAD/commit/74e329cc749069b4eef08c70218dde0d4231ae43 to be the cause.

@Rexbas I'm guessing there's something about the focal distance that's getting adjusted and making some vertexes in the Sketch vanish?

Rexbas, 2025-06-29:
That may be possible, I will have a look.

ErtugrulEge, 2025-06-29:
Similar problem https://github.com/FreeCAD/FreeCAD/issues/22086

Rexbas, 2025-06-29:
Ok, I did some testing. It looks like there is a problem that when the camera near and far distance are between ~1 and ~ -1. E.g. near = -0.257429 and far = -0.24925. The points are not correctly rendered. This also happens before merging https://github.com/FreeCAD/FreeCAD/commit/74e329cc749069b4eef08c70218dde0d4231ae43 but it is less likely to get this problem previously. Changing the `orthographicFocalDistance`  in https://github.com/FreeCAD/FreeCAD/commit/74e329cc749069b4eef08c70218dde0d4231ae43 to something like 5 reduces the probability of this problem in my experience. But that is no real solution.

Syres916, 2025-06-29:
> This also happens before merging [74e329c](https://github.com/FreeCAD/FreeCAD/commit/74e329cc749069b4eef08c70218dde0d4231ae43) but it is less likely to get this problem previously.

In the past I raised #19191 concerning https://github.com/FreeCAD/FreeCAD/commit/5969df37f4bd278798c13e2910acb8b96965a4a2 The conclusion was that the first sketch created (therefore no objects in the scene) was the most likely to have problems to the point where I create a 'hack' PR https://github.com/FreeCAD/FreeCAD/pull/19195

Rexbas, 2025-06-29:
Interesting, I will try to have a deeper look this week. If I cannot find a quick solution then I will just make a PR with `orthographicFocalDistance` = 5 or something probably next week.

Rexbas, 2025-07-05:
Note to myself, running this in the Python console makes it easier to reproduce and everything else except for the sketch must be hidden:
```
import FreeCADGui as Gui
from pivy import coin
view = Gui.ActiveDocument.ActiveView
camera = view.getCameraNode()
camera.position.setValue(10, 0, 10)
camera.focalDistance.setValue(1.0)
```

Syres916, 2025-07-15:
@bjnobre if you could please try the latest Weekly AppImage and report back if the vertices in Sketcher are visible and can be selected as you would expect, it would be appreciated, thanks.

closed 2025-07-16 by maxwxyz