FreeCAD/FreeCAD #22219: Ability to edit used sketch references during creation & modification of Part Design operations

techy-robot, 2025-06-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

Currently, when you do a basic part design operation like Pad, you select the sketch first, and you're done, there is no easy way to select specific edges to show up in a list to apply the pad to. Other CAD programs allow you to change the references in the creation menu, selecting sub sketches, using multiple sketches, etc... In order to do this in FreeCAD, you have to create the pad and go down into the properties to change the sketch reference and edges, and even then there is no list of what you selected.

In Realthunder's fork of FreeCAD he has implemented a feature that shows a list of all selected sketches, sketch parts, or edges, in the part creation and modification menu, a "Profile" option. This would be very helpful to have graphically.

### Full version info

```shell
Development Build weekly-2025.06.16
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

techy-robot, 2025-07-26:
What?? @maxwxyz why was this closed as a duplicate. This is not about the sketch subsections even though I mentioned it, it's about _the user interface of changing the selected sketch after creation_. Currently, FreeCAD allows you to select the edges of a sketch, or a sketch itself, but you cannot edit it without digging into properties to change the reference. It would be nice to have that list in the GUI operation popup. I should probably clarify the initial post.

RayHuber, 2026-04-30:
I agree that this is sorely needed now that "make internals" is an option. Once you create the pad/pocket/etc it is very unintuitive to change, esp if upstream steps break.

As an example, here you can see I open up that dialogue (which you have to already know about) to change the internals that should be used. While you are selecting them originally they show up with the SELECTED color per the preferences, but if you open the dialog again nothing is highlighted, which makes it quite confusing to make changes or know what is really currently selected.

It would be much better if either using this dialog or even opening the pad/pocket/etc commend highlighted the region (internals) to be used like when they are originally selected.

[image]