FreeCAD/FreeCAD #22220: I can not pocket at my condition

ChunGilLee, 2025-06-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

[pd_back_01.zip](https://github.com/user-attachments/files/20958250/pd_back_01.zip)

I attached my design file. I want to pocket with 'Sketch024'. 

[image]

My body disappear when I try to pocket the 'Sketch024'. And following log message is reported.

12:01:17  <TopoShape> TopoShapeExpansion.cpp(4608): makeElementFace: resulting face is invalid
12:01:17  Pocket011: Null shape
12:01:17  <TopoShape> TopoShapeExpansion.cpp(4608): makeElementFace: resulting face is invalid
12:01:17  Pocket011: Null shape
12:01:17  <TopoShape> TopoShapeExpansion.cpp(4608): makeElementFace: resulting face is invalid
12:01:21  <TopoShape> TopoShapeExpansion.cpp(4608): makeElementFace: resulting face is invalid

And, I tried to with some other options as this. You can see broken body.

[image]

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.1)
Hash: 878f0b8c9c72c6f215833a99f2762bc3a3cf2abd
Python 3.11.12, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

Katt-Kushol, 2025-06-28:
Hi, 
The sketch is not valid. Use of [Validate_Sketch](https://wiki.freecad.org/Sketcher_ValidateSketch) button shows problematic connections. 

[image]

It appears there is one too many constraints applied at the same location. 

[image]
It's surprising to see how 3-1.5mm constraints can be applied between same two edges but still have the sketch fully constrained, not over-constrained. Turns out there are overlapping edges in that location. 

[image]

Fixing the sketch may resolve the issue. Also, I recommend use of pattern tools available to accomplish the same goal with a simpler sketch.

ChunGilLee, 2025-06-29:
> Hi, The sketch is not valid. Use of [Validate_Sketch](https://wiki.freecad.org/Sketcher_ValidateSketch) button shows problematic connections.
>
> [image]
>
> It appears there is one too many constraints [...]

Thank you  for your reply. I will try your advice. But, I think that FreeCad should be report there are over-constraints and it should build right 3D object event if it has over-constraints or report something more useful information on the report window instead of showing the broken object.