FreeCAD/FreeCAD #22222: Segfault on Spreadsheet creation/recompute with Python 3.13 (works in 3.11)

KristjanMinn, 2025-06-28:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

## Summary

FreeCAD built with Python 3.13 segfaults when creating and recomputing a Spreadsheet object via macro. The same macro works fine with Python 3.11 (AppImage and other builds).

## Steps to reproduce

1. Start FreeCAD built with Python 3.13
2. Run this macro:

```python
import FreeCAD
doc = FreeCAD.ActiveDocument or FreeCAD.newDocument()
name = "MP"
# Remove old
for obj in doc.Objects:
    if obj.Name == name or obj.Label == name:
        doc.removeObject(obj.Name)
        break
# Create new
ss = doc.addObject("Spreadsheet::Sheet", name)
doc.recompute()

### Full version info

```shell
System information
OS: CachyOS (Hyprland/hyprland-uwsm/wayland)
Architecture: x86_64
FreeCAD version: 1.1.0devR42341 (Git shallow), build date: 2025/05/26
Python version: 3.13.0
Qt: 6.7.3
Coin: 4.0.3
Vtk: 9.3.1
boost: 1_86
Eigen3: 3.4.0
PySide: 6.7.3
shiboken: 6.7.3
xerces-c: 3.3.0
IfcOpenShell: 0.8.2
OCC: 7.8.1
Locale: English/United States (en_US)
Installed mods: AddonManager, BillOfMaterials, Woodworking-master, A2plus, ProDarkThemePreferencePack
Additional notes
The same macro works as expected with FreeCAD AppImage (Python 3.11).
This crash appears to be a core incompatibility with Python 3.13.

#Terminal output:
freecad  
FreeCAD 1.1.0, Libs: 1.1.0devR42341 (Git)
(C) 2001-2025 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Coin warning in glxglue_isdirect(): Couldn't get current GLX context.
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x44720) [0x7f54fc244720]
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-06-28:
I cannot reproduce the crash with Python 3.11 && 3.12.

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Unknown
Branch: master
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git) AppImage
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/95.8773/1
```

xtemp09, 2025-06-28:
@KristjanMinn, could you try to reproduce the crash with the latest AppImage?

KristjanMinn, 2025-06-28:
The latest appimage does not produce the crash and works as expected.
I am currently at:
OS: CachyOS (Hyprland/hyprland-uwsm/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/05/26 16:31:05
Build type: Release
Branch: main
Hash: bc5616dbf6cbe3a2d88992dc9572d6f5b5bb502f
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/140.257/1
Installed mods: 
  * AddonManager 2025.6.1
  * BillOfMaterials 1.0.3.2
  * Woodworking-master 1.0.1.39285.1.1.1.2025.06.11
  * A2plus 0.4.68
  * ProDarkThemePreferencePack 1.0.0
And this also does works as expected.
But the latest weekly build:
FreeCAD_weekly-builds-42133-conda-Linux-x86_64-py311.AppImage
Crashes out. 
So the error is introduced with recent changes.

xtemp09, 2025-06-28:
@KristjanMinn, sorry, I don't understand what you wrote. The latest weekly build is `FreeCAD_weekly-builds-42340-conda-Linux-x86_64-py311.AppImage`. [^1]

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

closed 2025-06-28 by KristjanMinn