FreeCAD/FreeCAD #22229: Replace uses of missing font family "Courier" with one that exists to avoid this cost.

pafech, 2025-06-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

At startup of the latest version of freeCAD 1.0.1 I receive the following error:

08:00:00  (qt.qpa.fonts) Populating font family aliases took 106 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 

This makes no sense to me since I don't use the font Courier. I guess there is some unused dependency that could be cleaned up.

### Full version info

```shell
There is no "About FreeCAD" dialog in this version. So I can't follow your instructions for this textfield here. However, the version is 1.0.1 macOS Silicon.
```

### Subproject(s) affected?

None

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-06-29:
Navigate to FreeCAD in the menu bar and then About FreeCAD and please copy info here

maxwxyz, 2025-06-29:
@hyarion you are on an arm Mac, any idea what is happening?

hyarion, 2025-06-29:
It does ring a bell, it might be a warning that doesn't really matter,  but I'll take a look at it.

xtemp09, 2025-06-29:
Duplicate of #12199.

maxwxyz, 2025-06-29:
Ah thanks!

closed 2025-06-29 by maxwxyz