FreeCAD/FreeCAD #22245: Workbench needs a freecad folder

leoheck, 2025-06-29:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

I am organizing a workbench repo and noticed that to have the code inside of a folder, the folder has to be lowercase `freecad`. This is a bit ugly solution. An ugly workround would be to place some files (Init.py, InitGui.py) in the root of the repository.

Is it possible somehow to place the code altogether on its own folder that is not lowercase `freecad`? 

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Core

### Anything else?

n/a

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

drwho495, 2025-06-30:
you can place the InitGui in the outermost folder, then have it import everything from an uppercase FreeCAD folder

leoheck, 2025-06-30:
Thanks @FlachyJoe. This link doesn't work.

leoheck, 2025-06-30:
The link started to work eventually. There is no mention on how shoul I nest the code inside the repository. 

@FlachyJoe do you have any repo example that has python code inside a folder which is not named `freecad`?

FlachyJoe, 2025-06-30:
Something like this should work
```<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<package format="1" xmlns="https://wiki.freecad.org/Package_Metadata">
    <name>…</name>
    <description>…</description>
    <version>…</version>
    <date>…</date>
    <maintainer email="">…</maintainer>
    <license file="LICENSE">…</license>
    <url type="repository" branch="main">https://github.com/…</url>
    <icon>…</icon>
    <content>
        <workbench>
            <name>…</name>
            <description>…</description>
            <classname>InitGui</classname>
            <subdirectory>./a/folder/somewhere/in/the/tree/</subdirectory>
            <icon>resources/icons/MyLogo.svg</icon>
            <tag>…</tag>
            <version>…</version>
        </workbench>
    </content>
</package>
```

The `InitGui` class should be search in `./a/folder/somewhere/in/the/tree/InitGui.py`
and the icon in `./a/folder/somewhere/in/the/tree/resources/icons/MyLogo.svg`