FreeCAD/FreeCAD #22247: Sketcher: The scale tool crashes the program or causes excessive constraints in the presence of non-scalable geometry.

kaiwas, 2025-06-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The scale tool crashes the program or causes excessive constraints in the presence of dimensional constraints and if the sketch contains fragments to which the scale does not apply.

Video demonstration.

https://github.com/user-attachments/assets/250496c8-356c-479e-ac1b-fd43e35ff92b

I was unable to get a crash in the video, but this happened very often in previous attempts.

A test file is attached. Remove the ZIP extension.

[Sketch_Scale_problem.FCStd.zip](https://github.com/user-attachments/files/20977068/Sketch_Scale_problem.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git) AppImage
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * addFC-main 1.6.2
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.63
  * freecad.gears 1.3.0
  * sheetmetal 0.7.23
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

maxwxyz, 2025-07-01:
Does this also crash in 1.0.1?
@AjinkyaDahale @PaddleStroke FYI

kaiwas, 2025-07-01:
> Does this also crash in 1.0.1?

no. in version 1.0.1 there are NO redundant constraints and crashes.

PaddleStroke, 2025-07-01:
So probably a regression. It sounds likely that it is from the auto-scale feature. Max do you know who wrote it?

luzpaz, 2025-07-01:
> So apparantly a regression. It sounds likely that it is from the auto-scale feature. Max do you know who wrote it?

Can someone ascertain where the regression stems from ?

xtemp09, 2025-07-01:
`git bisect` can pipoint the exact moment where the problem occurs. The issue here is that it does not happen consistently every time. @kaiwas, can you write a script to reproduce the problem reliably?

kaiwas, 2025-07-01:
> [@kaiwas](https://github.com/kaiwas), can you write a script to reproduce the problem reliably?

No. I don't know how to script at all)

Шопотом. Пользуясь случаем приглашаю вас в наш русскоязычный телеграм чат. https://t.me/freecad_russia

kaiwas, 2025-07-01:
This is probably a very similar problem. https://github.com/FreeCAD/FreeCAD/issues/22248

PR #23024 (2025-09-07): Sketcher: fix invalid constraint on first dimension

This PR deletes constraints before adding them back in sketcher's scale tool which is used by the autoscale feature. It also adds a way to skip solve on deletion of geometry/constraints which was necessary to fix a rendering bug. To do so a new enum DeleteOption has been created for deletion functions of SketchObject and some additional parameters have been exposed to the python API

## Issues

Fixes: #22954
Fixes: #22808
Fixes: #22248 
Fixes: #22247
closed 2025-09-07 by kadet1090