FreeCAD/FreeCAD #22248: Sketcher. The scale tool crashes the program or causes excessive constraints if we scale part of the object.

kaiwas, 2025-06-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Problem description

The scale tool crashes the program or causes excessive constraints in the presence of size constraints and if we scale part of the object.

Video demonstration.

https://github.com/user-attachments/assets/a6e4263c-3846-4a29-8245-f99630646242

I was unable to get a crash in the video, but this happened very often in previous attempts.

The test file is attached. Remove the ZIP extension.

[Sketch_Scale_problem_2.FCStd.zip](https://github.com/user-attachments/files/20977151/Sketch_Scale_problem_2.FCStd.zip)

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git) AppImage
Build date: 2025/06/27 09:25:36
Build type: Release
Branch: main
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Russian/Russia (ru_RU)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/
Logical DPI/Physical DPI/Pixel Ratio: 96/91.7938/1
Installed mods: 
  * addFC-main 1.6.2
  * Reinforcement
  * SvgWorkbench-main 1.0.0.dev5
  * PitchedRoof
  * LyapinWorkbrench
  * FreeCAD_SketchArch-master
  * Road 2025.2.22
  * PitchedRoof.zip
  * Curves 0.6.63
  * freecad.gears 1.3.0
  * sheetmetal 0.7.23
  * OpenTheme 2025.5.20
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [x] I agree to follow this project's Code of Conduct

xtemp09, 2025-06-30:
@tetektoza, the PR #21943 seems to cause the crash.

<details><summary>Backtrace</summary>

```cpp
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007ffff33ad4ca in QWidget::show() () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff33ad4ca in QWidget::show() ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007ffff5bce5e0 in Gui::EditableDatumLabel::setLockedAppearance (
    this=this@entry=0x55555aaeeae0, locked=locked@entry=true)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/EditableDatumLabel.cpp:533
#2  0x00007ffff5bce942 in Gui::EditableDatumLabel::eventFilter (this=0x55555aaeeae0, 
    watched=<optimized out>, event=0x7fffffffcc10)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/EditableDatumLabel.cpp:245
#3  0x00007ffff26d7e82 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff336bd35 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff33740e9 in QApplication::notify(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff5be4649 in Gui::GUIApplication::notify (this=0x7fffffffd3d0, 
    receiver=0x55555a8a4b40, event=0x7fffffffcc10)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplication.cpp:93
#7  0x00007ffff26d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff33ce00a in ?? () from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#9  0x00007ffff336bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#10 0x00007ffff5be4649 in Gui::GUIApplication::notify (this=0x7fffffffd3d0, 
    receiver=0x555557664590, event=0x7fffffffcc10)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/GuiApplication.cpp:93
#11 0x00007ffff26d8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#12 0x00007ffff2b41533 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#13 0x00007ffff2b17bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Gui.so.5
#14 0x00007fffec8fcd06 in ?? () from /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#15 0x00007ffff09145c5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff0973737 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff0913a63 in g_main_context_iteration ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff2735279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff26d6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff26df3e8 in QCoreApplication::exec() ()
   from /lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x00007ffff5b0ea18 in (anonymous namespace)::tryRunEventLoop (mainApp=...)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2238
#22 0x00007ffff5b0ed0f in (anonymous namespace)::runEventLoop (mainApp=...)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2267
#23 0x00007ffff5b12c21 in Gui::Application::runApplication ()
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Gui/Application.cpp:2347
#24 0x000055555556a9ad in main (argc=<optimized out>, argv=<optimized out>)
    at /home/xtemp09/Downloads/FreeCAD/freecad-source/src/Main/MainGui.cpp:321
(gdb)
``` 

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42340 (Git)
Build date: 2025/06/27 09:25:36
Build type: Unknown
Branch: master
Hash: 4f167b828b68b1544732881f0b2393ea873cffd1
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/unset/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

</details>

maxwxyz, 2025-07-01:
@AjinkyaDahale @PaddleStroke FYI

chennes, 2025-07-26:
I think really @theo-vt is the one to ping here -- The code that's trying to figure out whether to create new constraints when scaling is incorrectly determining that ALL sketch constraints, even those in unaffected bodies, should be copied.

PR #23024 (2025-09-07): Sketcher: fix invalid constraint on first dimension

This PR deletes constraints before adding them back in sketcher's scale tool which is used by the autoscale feature. It also adds a way to skip solve on deletion of geometry/constraints which was necessary to fix a rendering bug. To do so a new enum DeleteOption has been created for deletion functions of SketchObject and some additional parameters have been exposed to the python API

## Issues

Fixes: #22954
Fixes: #22808
Fixes: #22248 
Fixes: #22247
closed 2025-09-07 by kadet1090